paulober/MicroPico #118: No connection to Pico from raspberry pi 400

pjleeflyer, 2023-07-28:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. flashed pico with rp2-pico-20230426-v1.20.0.uf2
2. disconnected and re-connected pico to raspberry pi 400
3. Added user pi to group dialout
4. changed group on /dev/ttyACM0 to dialout
5.  Run ms-Visual Studio Code

## What happens?
pico remains disconnected with error message:
"Error while reading (COM) ports for autoConnect: Invalid response"

## What were you expecting to happen?
pico to connect

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

"Error while reading (COM) ports for autoConnect: Invalid response"

## Any other comments?

Note - I am using a regular pico and NOT the pico-w. Does this matter? I have scoured the internet for solutions but non found.

## Which version of Pico-W-Go are you using?

v3.2.0

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.76.0
Commit: 92da9481c0904c6adfe372c12da3b7748d74bdcb
Date: 2023-03-01T10:25:14.463Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Linux arm 5.10.103-v7l+
Sandboxed: No
```

paulober, 2023-08-01:
On RaspberryPi OS you don't have to add pi to a group or modify any permissions by default to access the serial port.
try:
```bash
cd ~/.vscode/extensions/paulober.pico-w-go-3.2.0-linux-armhf/dist/scripts
python3 ./wrapper.py --scan-ports
```
what output do you get?

pjleeflyer, 2023-08-03:
Thanks for the help! - I received the following error:

pi@raspberrypi:~ $ cd ~/.vscode/extensions/paulober.pico-w-go-3.2.0-linux-armhf/dist/scripts
pi@raspberrypi:~/.vscode/extensions/paulober.pico-w-go-3.2.0-linux-armhf/dist/scripts $ python3 ./wrapper.py --scan-ports

Traceback (most recent call last):
  File "./wrapper.py", line 8, in <module>
    from utils import create_folder_structure, wrap_expressions_with_print, prepend_parent_directories
  File "/home/pi/.vscode/extensions/paulober.pico-w-go-3.2.0-linux-armhf/dist/scripts/utils.py", line 6, in <module>
    def create_folder_structure(file_paths: list[str], local_folder_path: str):
TypeError: 'type' object is not subscriptable

pjleeflyer, 2023-08-03:
I think I'm using the wrong version of python3 ???

pi@raspberrypi:~/.vscode/extensions $ python --version
Python 3.10.12
pi@raspberrypi:~/.vscode/extensions $ python3 --version
Python 3.7.3

pjleeflyer, 2023-08-03:
I corrected the version of Python3 to Python 3.10.12. Now I get this:

$ python3 ./wrapper.py --scan-ports
!!ImportError!!
Please install pyserial

However I have already installed pyserial.... ($ pip install pyserial)

paulober, 2023-08-03:
> I think I'm using the wrong version of python3 ???
> 
> 
> 
> pi@raspberrypi:~/.vscode/extensions $ python --version
> 
> Python 3.10.12
> 
> pi@raspberrypi:~/.vscode/extensions $ python3 --version
> 
> Python 3.7.3

Yes, the extension has minimum Python 3.9 in its requirements. That's the current Python version from the Raspberry Pi OS repositories.

paulober, 2023-08-03:
> I corrected the version of Python3 to Python 3.10.12. Now I get this:
> 
> 
> 
> $ python3 ./wrapper.py --scan-ports
> 
> !!ImportError!!
> 
> Please install pyserial
> 
> 
> 
> However I have already installed pyserial.... ($ pip install pyserial)
> 
> 

Try `python3 -m pip install pyserial`...

pjleeflyer, 2023-08-04:
Still get the same response:

$ sudo python3 -m pip install pyserial
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (3.4)

pi@raspberrypi:~/.vscode/extensions/paulober.pico-w-go-3.2.0-linux-armhf/dist/scripts $ python3 ./wrapper.py --scan-ports
!!ImportError!!
Please install pyserial

pjleeflyer, 2023-08-04:
On startup of Visual Studio Code I now get this error:

[error] [Extension Host] [Pico-W-Go] Failed to install pyserial:  Error: Command failed: /usr/local/bin/python3.10 -m pip install --upgrade pyserial
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.

paulober, 2023-08-04:
Do not install pip package as sudo, that's bad practice and restricts the access of python scripts to these packages if they are not run as root.
```bash
sudo python3 -m pip uninstall pyserial
python3 -m pip install -U pip
python3 -m pip install pyserial
```

Also there seem to be a problem with you python installation concerning the ssl module. Do you use the python from `sudo apt install python3`?

KazW, 2023-08-11:
I had the same problem on Arch Linux trying to get this extension to work. The wrapper script was able to pick up on the com ports, but the extension still didn't work... Too bad, would have been nice using vscode.

I tried this with a pico and a pico w. The only odd thing about my setup is that my python install is done through `asdf`, and I had tried to setup a venv for the project using pipenv, but also had the same error with no venv.

paulober, 2023-08-11:
@KazW The extension uses your python3 executable from the path and installs `pyserial` with `python3 -m pip install pyserial`. The python scripts are then run with `python3 ./wrapper.py` where `pyserial` should then be available. 
The python executable can be changed in settings. If the wrapper script don't pick up your connected picos this might be due to the Linux permission issue I mentioned in the wiki: https://github.com/paulober/MicroPico/wiki/Linux

KazW, 2023-08-11:
@paulober It wasn't a permisson issue, I already had my user as part of the `uucp` group (equivalent to `dialout` on Debian). I use various TTL chips along with various ESP/Arduino boards in my projects, so I setup permissions when I installed Arch, as well as creating udev rules for other non-serial devices I use.

But for some reason, it did start working once I installed `pyserial` via the system package manager. So if I had to make a guess, vscode or this extension probably don't use the right `python3` in my `PATH`. It is odd though, vscode uses `asdf` pythons when I use it on my work computer with Flask/Django projects with that use a venv.

paulober, 2023-08-14:
@KazW Ok so that asdf managed python installation is probably on top in you PATH (or at least the PATH VS Code is launched with): `which python3`.

paulober, 2023-08-14:
@pjleeflyer Do you still have the issue?

paulober, 2023-08-20:
Since I haven't heard back from you I assume the issue is now resolved.

closed 2023-08-20 by paulober