paulober/MicroPico #125: machine.Pin not recognized

skeidr, 2023-08-15:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Micropython code:
import machine
led_onboard = machine.Pin('LED', machine.Pin.OUT) 

## What happens?
Pylance highlights Pin with message:
Pin is not a known member of module machine

## Any other comments?
Working fixes:
Use import umachine in stead of import machine
or go back one version from v. 3.2.2 to v. 3.2.1
Seems to be a stub definition/import problem between machine and umachine

## Which version of MicroPico are you using?
3.11.4 64-bit

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.81.1 (user setup)
Commit: 6c3e3dba23e8fadc360aed75ce363ba185c49794
Date: 2023-08-09T22:22:42.175Z
Electron: 22.3.18
ElectronBuildId: 22689846
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.22621
```

skeidr, 2023-08-15:
My MicroPico version: 3.2.2

ithatguy, 2023-08-15:
seems to be picking up the module at .vscode/Pico-W-Stub/stubs/machine/machine.pyi

paulober, 2023-08-15:
@skeidr Thanks for reporting this issue and I'm sorry for the inconvenience.
@ithatguy Thanks for helping out.

I just published a hotfix (v3.2.3) to resolve this issue.

ithatguy, 2023-08-15:
Additionally, I had to change this in a few more places. For example: MicroPico/stubs/stdlib/sys.pyi

paulober, 2023-08-15:
> Additionally, I had to change this in a few more places. For example: MicroPico/stubs/stdlib/sys.pyi

I also fixed these relative imports...

commit 76e74563 (2023-08-15): Stubs updated to patch v1.20.0-4

-Fixes #125
PR #126 (2023-08-15): Hotfix v3.2.3
### Changed
- Updated stubs to patch v1.20.0-4 stable (Fixes #125)