paulober/MicroPico #128: Can this work with Circuitpython

jevers76, 2023-08-16:
# Feature Request 🚀
I tried using with a Pico running Circuitpython but was not able to connect.  The message was No COM device found.  Please check your connection or ports and try again.  Is it possible to use this with Circuitpython or did I do something wrong?

paulober, 2023-08-20:
The reason why it's not detected is because a Pico running circuitpython has a different CNC product id compared to a Pico running MicroPython.
The extension relies on certain modules to be available on the Pico and certain ascii commands to work in the repl.
Also currently the extension only contains stubs for MicroPython.

So you see there is all lot work that needs to be done in order for Circuitpython boards to work with the extension.
I'll take a look on how much time it would take...

paulober, 2023-08-21:
The detection discrepancy arises due to distinct CNC product IDs between Pico boards using CircuitPython and MicroPython. The extension's functionality relies on specific Pico modules and ASCII commands within the REPL. Presently, the extension exclusively offers MicroPython stubs. Thus, substantial effort is required to enable compatibility with CircuitPython boards. I will assess the timeframe necessary for this implementation and see if this is something I would be able to add to the extension's functionality.

paulober, 2026-09-14:
Thanks for the suggestion! MicroPico focuses on MicroPython, and CircuitPython works quite differently (its USB drive workflow and REPL), so I'm closing this as out of scope.

closed 2026-09-14 by paulober