paulober/MicroPico #129: Bluetooth support for auto-completion

montanag, 2023-08-29:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.
When writing bluetooth code for the pico w, the bluetooth module is not recognized by the extension.
[image]

## Describe the solution you'd like
Ideally, the import and it's methods etc would be recognized.

## Describe alternatives you've considered
Considered trying to write out a stub file in the `.vscode/Pico-W-Stub/stubs/` directory.

## Teachability, Documentation, Adoption, Migration Strategy
As mentioned, the auto-complete is based off of [rp2-pico-w-20230426-v1.20.0.uf2](https://micropython.org/resources/firmware/rp2-pico-w-20230426-v1.20.0.uf2). This firmware doesn't seem to support bluetooth connectivity like [this one](https://datasheets.raspberrypi.com/soft/micropython-firmware-pico-w-130623.uf2) does.

paulober, 2023-08-29:
Will be updated in an upcoming release.

montanag, 2023-08-31:
Fantastic! Any issues or anything I can subscribe to other than this one? Will it also support libraries like `aioble`? Thanks!

paulober, 2023-09-05:
The extension is now updated to the current beta firmware. Notice that the auto-completion for bluetooth and aioble are still in an unstable state like the firmware. They can change from day to day or there are errors. If you find a bug please report them in the [Pico-W-Stub](https://GitHub.com/paulober/Pico-W-Stub) repo.

closed 2023-09-05 by paulober

montanag, 2023-09-05:
Wow, very fast turn around on this. Thank you, and can't wait to play around with the new update