paulober/MicroPico #133: The MicroPico stuff doesn't respond or appear in the status bar anymore.

jms1989, 2023-09-17:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Install Extension
2. Go to pico project workspace

## What happens?

Nothing, not a thing. Clicking a micropico command does nothing. 

## What were you expecting to happen?

The extension to respond and appear in the status bar so I can run my code

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

https://gist.github.com/jms1989/7913ea59c8d23030485a88e40ce053d6

## Any other comments?

## Which version of MicroPico are you using?
MicroPico v3.2.5

## Support info

```
Version: 1.82.2
Commit: abd2f3db4bdb28f9e95536dfa84d8479f1eb312d
Date: 2023-09-14T05:51:20.981Z
Electron: 25.8.1
ElectronBuildId: 23779380
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux x64 6.2.0-32-generic
```

xabadabazinho, 2023-09-19:
Hi, had the same issue here. 
Installed a previos version,3.0. 
Run "Configure Project": the extension asks about pyserial and I agreed to install it.
Reinstall current version.
Quit VSCode and delete everything inside the folder.
Now its working.

jms1989, 2023-09-22:
Downgraded the plugin to v3.2.4. It's working again. Not sure what broke.

junaidcodingmaster, 2023-09-24:
> Nothing, not a thing. Clicking a micropico command does nothing.

Same Issue

junaidcodingmaster, 2023-09-24:
I tried fixing things by downgrading a plugin to v3.2.4, and for a moment, it seemed like it was back to normal. But then, that same old error came back. Honestly, I'm stumped because I can't figure out what caused it in the first place.

In my quest for a solution, I took things a step further and downgraded the plugin even more, down to v2.1.8. Unfortunately, that didn't do the trick either. I've been trying different versions and fiddling around, but that stubborn error just won't budge.

The error message I keep seeing is: 
- `ERROR: microPico:init command not found.`
- `Please connect your device` even though my pico is connected and clearly visible in the Device Manager.

As for my computer, here are the specs:

- Visual Studio Code Version: 1.82.2 (user setup)
- Commit: abd2f3db4bdb28f9e95536dfa84d8479f1eb312d
- Date: 2023-09-14T05:55:25.390Z
- Electron: 25.8.1
- ElectronBuildId: 23779380
- Chromium: 114.0.5735.289
- Node.js: 18.15.0
- V8: 11.4.183.29-electron.0
- OS: Windows_NT x64

If you've got any ideas on how I can fix this, I'd really appreciate your help! . ok, byeeee👋

contact me 
- [ junaid@abujuni.dev ]
- [ [Junaid](https://abujuni.dev) ]

paulober, 2023-09-27:
> ## Any logs, error output, etc?
> _(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go)_ _(If it’s long, please paste to https://gist.github.com and insert the link here)_
> 
> https://gist.github.com/jms1989/7913ea59c8d23030485a88e40ce053d6
> 

I seems like you've set a venv somewhere so the vscode Python extension tell MicroPico to use it though it doesn't exist anymore or the extension is unable to use it for installing the pyserial pip package.
You can check your PATH for all python executables in there and if you selected a venv in VSCode in the Python extension.

paulober, 2023-10-10:
Should be fixed by update [`v3.3.0`](https://github.com/paulober/MicroPico/releases/tag/v3.3.0) as it removes the dependency on your local python installation/environment.

paulober, 2023-10-10:
@jms1989 @junaidcodingmaster @xabadabazinho Can you confirm that the issue is also resolved for you?

closed 2023-10-10 by paulober

xabadabazinho, 2023-10-11:
> @jms1989 @junaidcodingmaster @xabadabazinho Can you confirm that the issue is also resolved for you?
Solved, thanks.