paulober/MicroPico #140: Installing/Updating MicroPico extension in VS Code Triggers Windows Security AntiVirus Warning.

erzahurak, 2023-10-11:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Install or Update MicroPico Extension to the latest version of VSCode OR open a micropython project folder with the newest extension version installed.

or:

2. You can verify by uninstalling MicroPico extension, restarting VSCode, and then re-installing MicroPico extension as well.

## What happens?

The following Windows Security notification is displayed:

[image]

> WIndows Security
>
> Threats found
> Microsoft Defender Antivirus found threats. Get Details

Clicking on the notification opens up the WIndows Security settings panel.  You can navigate to the threat details and it shows:

[image]

>Threat Quarantined
>(date/time of action)
>
>Detected: Trojan:Win32/Wacatac.H!ml
>Status: Quarantined
>Quarentined files are in a restricted area where they can't harm your device.  They will be removed automatically.
>
>Date: (date/time of action)
>Details: This program is dangerous and executes commands from an attacker.
>
>Affected items:
>file: (path to user)\AppData\Local\Temp\onefile_onefile_19324_133414634768884204\wrapper_windows-latest_amd64.exe
> file: (path to user)AppData\Local\Temp\onefile_19324_133414634768884204\wrapper_windows-latest_amd64.exe

Clicking learn more jumps to web page about the threat:
https://go.microsoft.com/fwlink/?linkid=142185&name=Trojan:Win32/Wacatac.H!ml&threatid=2147814523

--

Also, quite logically, the MicroPico extension doesn't work, so you can't do anything with MicroPico in VSCode. :(

## What were you expecting to happen?

Absolutely nothing out of the ordinary.  Was just hoping to do some micropython coding.

## Any logs, error output, etc?

nothing but this, and the info above:

'''
2023-10-10 22:04:36.646 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'workspaceContains:.micropico'
'''

## Any other comments?

Thanks for looking into this!

## Which version of MicroPico are you using?

MicroPico
v3.3.1
Published: 2022-8-10, 10:43:54
Last released: 2023-10-10, 19:38:39
Identifier: paulober.pico-w-go

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.83.0 (user setup)
Commit: e7e037083ff4455cf320e344325dacb480062c3c
Date: 2023-10-03T16:12:16.321Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22621
```

LShortcuts, 2023-10-11:
I am also experiencing this issue.

m4verick84, 2023-10-11:
can confirm, got the same error.

paulober, 2023-10-11:
Thanks for the fast report. I'll contact Microsoft on this false positive. It could help if you could add your Defender Security Definitions version.

LShortcuts, 2023-10-11:
1.399.407.0 is my definition version

paulober, 2023-10-11:
Thanks. Let's see how fast Microsoft can resolve this. You'll need to update your Defender Security Definitions after the analysis finished. I'll keep this thread updated...

LShortcuts, 2023-10-11:
Here's the VirusTotal report: https://www.virustotal.com/gui/file/dc461764a1ea0f3dc806ed56300587417196027b71673ea58605e91eca6323ac/detection

erzahurak, 2023-10-11:
Paul, thanks for the fast attention!  Updated my Windows Security signatures, and I'm back in business!
LShortcuts, thanks for the contributing additional information!

Cheers!

LoreSchaeffer, 2023-10-12:
Hi, I have the same problem with Bitdefender, the file wrapper_win32_amd64.exe is marked as Trojan.GenericKD.69714177. I tried adding it as an exception but the file is deleted anyway and the antivirus also require a reboot of the system to "clean the system from all the threats"

paulober, 2023-10-12:
@LoreSchaeffer Thanks for reporting, I'll take care of it.