paulober/MicroPico #142: Same issue as #140

mickcleri, 2023-10-11:
Hi Paul,

First of all thx a lot for your work on MicroPico.

I, unfortunately, face the same issue as #140 
but my def are UTD i believe :

Service Version: 4.18.23090.2008
Engine Version: 1.1.23090.2007
AntiSpyware Signature Version: 1.399.447.0
AntiVirus Signature Version: 1.399.447.0

file: C:\Users\XXXXxxxXXXX\AppData\Local\Temp\onefile_7628_133415356419041280\wrapper_windows-latest_amd64.exe

- Pico pi not getting detected as a result of the false positive in Wind Defender

-Tried to allow the *Threat*, doesn't solve it

Could you double check?

Regards,

Mick

erzahurak, 2023-10-12:
Confirming, I'm seeing the same thing.  Different threat signature this time.  I have a feeling we're in for a bit of whack-a-mole here. :(

This one is a little more incessant, though.  For some reason it keeps re-writing versions of itself to the temp folder.  14K instances (about 128GB!) since about 11am local time, until I uninstalled the the MicroPice extension again.

[image]

The contents of all these folders is identical:

[image]

Threat Details in Windows Security:

```
Detected: Program:Win32/Wacapew.C!ml
Status: Quarantined

Affected items:
file: C:\Users\XXXXXXX\AppData\Local\Temp\onefile_12468_133415484931835117\wrapper_windows-latest_amd64.exe
```

Windows Security threat signatures stuff:

```
>cd c:\Program Files\Windows Defender

>MpCmdRun.exe -removedefinitions -dynamicsignatures

Service Version: 4.18.23090.2008
Engine Version: 1.1.23090.2007
AntiSpyware Signature Version: 1.399.447.0
AntiVirus Signature Version: 1.399.447.0

Starting Dynamic Signature removal.
Done!

Service Version: 4.18.23090.2008
Engine Version: 1.1.23090.2007
AntiSpyware Signature Version: 1.399.447.0
AntiVirus Signature Version: 1.399.447.0

>MpCmdRun.exe -SignatureUpdate

Signature update started . . .
Service Version: 4.18.23090.2008
Engine Version: 1.1.23090.2007
AntiSpyware Signature Version: 1.399.447.0
AntiVirus Signature Version: 1.399.447.0
Signature update finished.
```

erzahurak, 2023-10-12:
Found this in an old reddit thread.  Posting in case the information is helpful.  If I'm insulting anyone's intelligence, that is not my intent, please forgive me. :)

https://www.reddit.com/r/computerviruses/comments/mkrmg7/windows_defender_found_wacapewc_ml/h4517q7/

> It is not necessarily a Trojan.
> 
> It is a component of PyInstaller, a Python tool that converts a Python script into an executable for computers that don't have python installed, or where you just want to double click and run. If the developer used the flag --uac-admin, PyInstaller adds wacapew.c to the executable to request the user receive a UAC Prompt to request Admin privileges.

mickcleri, 2023-10-12:
> If I'm insulting anyone's intelligence

I was dumb enough to run a full AV scared of getting a ransomware, till i launched VScode again.... :D

erzahurak, 2023-10-12:
> I was dumb enough to run a full AV scared of getting a ransomware, till i launched VScode again.... :D

I went into full-on panic mode yesterday before I figured out what was going on.  I'd say that's just a natural -- and smart -- reaction. :)

erzahurak, 2023-10-12:
@mickcleri , @paulober,

Update:  As of Windows Security security intelligence update 1.399.487.0, 10/12/2023 3:46AM, my PC is no longer freaking out about the MicroPico extension on install. :)  @mickcleri , can you grab the latest update and confirm?

mickcleri, 2023-10-12:
@erzahurak, @paulober 

Great news indeed, after last def update no more False Positive, closing.

Service Version: 4.18.23090.2008
Engine Version: 1.1.23090.2007
AntiSpyware Signature Version: 1.399.482.0
AntiVirus Signature Version: 1.399.482.0

Thanks a lot ! ( i hate Thonny )

closed 2023-10-12 by mickcleri