paulober/MicroPico #143: Similar to #140.  Continuous creation of new "onefile_xxx"folders while MicroPico is enabled

DrPeterRobinson, 2023-10-13:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Install MicroPico and enable it

## What happens?

Every few seconds, a new folder of the form "onefile_1760_133416617039058651" (with unique numbers) is silently created in the user's Appdata/Temp folder.  This continues until there is no more available space on the disk.  No warnings or error messages are displayed.  Each folder contains the same files:

```
    Directory: C:\Users\squis\AppData\Local\Temp\onefile_4536_133416633069551506
Mode                 LastWriteTime         Length Name                                                                 
----                 -------------         ------ ----                                                                 
-a----        13/10/2023     10:35          39696 libffi-8.dll                                                         
-a----        13/10/2023     10:35        5765912 python311.dll                                                        
-a----        13/10/2023     10:35          30488 select.pyd                                                           
-a----        13/10/2023     10:35        1122768 ucrtbase.dll                                                         
-a----        13/10/2023     10:35        1141016 unicodedata.pyd                                                      
-a----        13/10/2023     10:35         109440 vcruntime140.dll                                                     
-a----        13/10/2023     10:35        6222336 wrapper_windows-latest_amd64.exe                                     
-a----        13/10/2023     10:35          84760 _bz2.pyd                                                             
-a----        13/10/2023     10:35         123672 _ctypes.pyd                                                          
-a----        13/10/2023     10:35         159000 _lzma.pyd                                                            
-a----        13/10/2023     10:35          79640 _socket.pyd                                                          
```

Disabling the MicroPico extension stops the creation of new folder.  Enabling it starts it again.

I am using Windows 10 version 10.0.19045 build 19045

## What were you expecting to happen?

Not this :-)

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

No logging messages relating to this issue are produced in the terminal.  Windows system logs do not show anything.  Windows defender logs so not show anything.

## Any other comments?

I have updated Visual Studio Code and Windows Defender definitions as recommended in #140 and #142.  I have uninstalled and reinstalled the latest MicroPico extension.  At no time have I seen warning virus or trojan messages as other users seem to have.  The first indication of a problem is a warning that there is no more space on the disk, at which point investigation discovers thousands of 'onefile' folders taking up all the previously available space.

MicroPico is otherwise working fine and is a hugely useful tool!

## Which version of MicroPico are you using?

3.3.1

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.83.1 (system setup)
Commit: f1b07bd25dfad64b0167beb15359ae573aecd2cc
Date: 2023-10-10T23:48:05.904Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.19045
```

bobbysq, 2023-10-14:
I'm also having this issue after updating VSCode. I'm pretty sure it hadn't been affecting me before I did that, based on how quickly it seems to generate the onefile folders. Here's my support info:

```
Version: 1.83.1 (system setup)
Commit: f1b07bd25dfad64b0167beb15359ae573aecd2cc
Date: 2023-10-10T23:48:05.904Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22621
```

erzahurak, 2023-10-15:
Confirmed, I'm seeing the same thing.  Came back from the workshop to a 0 bytes free C: drive. :)

erzahurak, 2023-10-15:
The onefile_xxx folders contain a file named: wrapper_windows-latest_amd64.exe

Could this be related to the recent updates removing arm support?

Just guessing, I have nooooo idea how publishing/releasing one of these extensions works.

erzahurak, 2023-10-15:
Is this related to pyboard-serial-com's pyrunner/wrapper stuff?

It's almost like it's pulling the precompiled bin into temp any time it hits an `import { PyboardRunner } from "@paulober/pyboard-serial-com";` or something.

(This is an assumption from looking at the pyboard-serial-com repo, not from running any tests.)

EduardoOliveira, 2023-10-16:
Same behavior here

```
Version: 1.83.1 (user setup)
Commit: f1b07bd25dfad64b0167beb15359ae573aecd2cc
Date: 2023-10-10T23:48:05.904Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.19044
```

closed 2023-10-16 by paulober

erzahurak, 2023-10-16:
Thank you, Paul!

DrPeterRobinson, 2023-10-17:
I can confirm that the problem has been fixed.  Thanks for responding so quickly, Paul.  

MicroPico is a really useful add-on.  I was so keen to keep using it that I wrote a script to continually delete the rogue files in the background.  I glad I don't need that any more :-)

repentsinner, 2025-08-15:
I'm still seeing this, in VS Code 1.103.1 on Windows ARM64.

I installed the extension from the marketplace and it says it's version 3.3.0, not 4.2.2, although the marketplace seems to be aware that 4.2.2 exists and the latest release date coincides with the 4.2.2 release date.

[image]

If I select "Download Specific Version VSIX", up to 4.2.2 is available.

[image]

```
Version: 1.103.1 (user setup)
Commit: 360a4e4fd251bfce169a4ddf857c7d25d1ad40da
Date: 2025-08-12T16:25:40.542Z
Electron: 37.2.3
ElectronBuildId: 12035395
Chromium: 138.0.7204.100
Node.js: 22.17.0
V8: 13.8.500258-electron.0
OS: Windows_NT arm64 10.0.26100
```

Is this an ARM64 issue?

repentsinner, 2025-08-15:
Oh, I see the issue. On non-Windows ARM64 systems, the most recent VSIX (?) loads in the marketplace, and shows that Windows ARM64 is unsupported. I guess it _was_ supported until 3.3.0 and then support stopped, but then it hasn't been marked as deprecated (https://github.com/microsoft/vscode-discussions/discussions/1).

PR #145 (2023-10-16): Hotfix v3.3.2
### Changed
- Updated dependencies
- Upgraded to `pyboard-serial-com` `v2.0.1`
- Fixes Windows issues related to update `v3.3.0` (Fixes #143)