paulober/MicroPico #144: Extension crashes when select configure project

shengdoescoding, 2023-10-15:
I'm facing this problem.

See detail below:

## What are the steps to reproduce this issue

1. Run `>MicroPico: Configure project` from VSCode code command Pallet

## What happens?

Same as #65, only the activating extensions at the bottom bar appears for a few seconds and then nothing.

## WHat were you expecting to happen?

For the extension to start and configure the directory as micro pico project.

## Any logs, error output, etc?

2023-10-15 16:17:31.595 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onCommand:micropico.initialise'

## Any other comments?

I have tried most of the suggestions from #65.

I have made sure that I have Pyserial installed.

I tried to edit settings but pythonPath is no longer an acceptable config item.

I have installed Python 3.11.

I have satisfied the requirement of havingMicroPython flashed onto the Rasberry Pi Pico

I can turn the LED on using minicom

`/usr/local/bin` is in `$PATH`, and python executable is at `/usr/bin/python3.11` along with a couple other python versions

## Support info

```
Version: 1.83.1
Commit: f1b07bd25dfad64b0167beb15359ae573aecd2cc
Date: 2023-10-10T23:45:31.402Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux x64 5.15.0-86-generic
```

Happy to provide any more information I have missed

paulober, 2023-10-16:
Since `v3.3.0` you don't need to install Python or pyserial anymore.
Have you opened a folder before running `Configure Project` command?

closed 2023-10-19 by paulober

choncba, 2023-11-10:
Hi, I have exactly the same problem with the latest version 3.4.2.
I use MX Linux (Debian 11) and Python 3.9.2. 

VSCode:

`Version: 1.84.1
Commit: 2b35e1e6d88f1ce073683991d1eff5284a32690f
Date: 2023-11-06T12:35:31.181Z
Electron: 25.9.2
ElectronBuildId: 24603566
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux x64 6.0.0-6mx-amd64`

Extension Logs:
`2023-11-10 01:02:34.884 [info] Extension host with pid 9934 started
2023-11-10 01:02:34.884 [info] Skipping acquiring lock for /home/lbono/.config/Code/User/workspaceStorage/a3a479e8e3316a542340f746a49724c3.
2023-11-10 01:02:35.127 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-11-10 01:02:35.144 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python'
2023-11-10 01:02:35.321 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2023-11-10 01:02:35.462 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2023-11-10 01:02:35.986 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-11-10 01:02:36.292 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-11-10 01:02:37.499 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-11-10 01:02:37.547 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-11-10 01:02:37.700 [info] Eager extensions activated
2023-11-10 01:02:37.703 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-11-10 01:02:37.890 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-11-10 01:02:37.896 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-11-10 01:02:51.126 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onCommand:micropico.initialise'`

Any idea?

Regards.-

paulober, 2023-11-10:
@choncba Hi,thanks for reporting. What output do you get for following commands with your Pico connected?

```
cd ~/.vscode/extensions/paulober.pico-w-go-3.4.2-linux-x64/dist/scripts/wrapper_linux_amd64/
./wrapper_linux_amd64.bin --scan-ports
```
And
```
ldd --version
```

choncba, 2023-11-10:
> @choncba Hi,thanks for reporting. What output do you get for following commands with your Pico connected?
> 
> ```
> cd ~/.vscode/extensions/paulober.pico-w-go-3.4.2-linux-x64/dist/scripts/wrapper_linux_amd64/
> ./wrapper_linux_amd64.bin --scan-ports
> ```
> 
> And
> 
> ```
> ldd --version
> ```

@paulober thanks for the response, apparently GLIBC_2.35 in my distribution is missing :
```
./wrapper_linux_amd64.bin --scan-ports
./wrapper_linux_amd64.bin: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./wrapper_linux_amd64.bin)
./wrapper_linux_amd64.bin: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./wrapper_linux_amd64.bin)
./wrapper_linux_amd64.bin: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /home/lbono/.vscode/extensions/paulober.pico-w-go-3.4.2-linux-x64/dist/scripts/wrapper_linux_amd64/libpython3.11.so.1.0)
./wrapper_linux_amd64.bin: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/lbono/.vscode/extensions/paulober.pico-w-go-3.4.2-linux-x64/dist/scripts/wrapper_linux_amd64/libpython3.11.so.1.0)
./wrapper_linux_amd64.bin: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/lbono/.vscode/extensions/paulober.pico-w-go-3.4.2-linux-x64/dist/scripts/wrapper_linux_amd64/libpython3.11.so.1.0)
./wrapper_linux_amd64.bin: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/lbono/.vscode/extensions/paulober.pico-w-go-3.4.2-linux-x64/dist/scripts/wrapper_linux_amd64/libpython3.11.so.1.0)
```
```ldd --version``` looks ok, but I have an older version than required (spanish response, sorry):
```
ldd --version
ldd (Debian GLIBC 2.31-13+deb11u7) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
Esto es software libre; vea el código fuente para las condiciones de copia.
No hay NINGUNA garantía; ni siquiera de COMERCIABILIDAD o IDONEIDAD PARA UN
FIN DETERMINADO.
Escrito por Roland McGrath y Ulrich Drepper.
```
Can't found a secure method to upgrade it.

Regards.-

paulober, 2023-11-10:
@choncba Thanks for all the information. This issue, as I guessed, is the GLIBC backwards-compatibility. 
I know, that's not a good idea.
I'll release a patch in the next hour which adds backwards-compatibility so it'll run on your system.

choncba, 2023-11-10:
> @choncba Thanks for all the information. This issue, as I guessed, is the GLIBC backwards-compatibility. I know, that's not a good idea. I'll release a patch in the next hour which adds backwards-compatibility so it'll run on your system.

Man, you're awesome, thanks for your great job.

paulober, 2023-11-10:
> Man, you're awesome, thanks for your great job.

Thanks! Patch will be distributed in the VS Code Marketplace within the next hour: (v3.4.3 release)[https://github.com/paulober/MicroPico/releases/tag/v3.4.3]

choncba, 2023-11-10:
> > Man, you're awesome, thanks for your great job.
> 
> Thanks! Patch will be distributed in the VS Code Marketplace within the next hour: [v3.4.3 release](https://github.com/paulober/MicroPico/releases/tag/v3.4.3)

Upgraded and Working!
[image]
Tanks a lot!

PR #152 (2023-10-19): Patch v3.4.1
### Changed
- Upgraded to `pyboard-serial-com` `v2.0.3` (Fixes #144)
- Updated stubs to MicroPython v1.21.0 stable
- Updated dependencies
- Adds back linux `armhf` support