paulober/MicroPico #153: Autorun main.py script fails almost immediately if connected to MicroPico

WhatNot911, 2023-10-20:
> Please provide a clear and concise description of the bug along with logs
Script uploaded to the board stops **almost immediately** after a **hard reset**/usb unplug if vREPL is running in the VSCode

## What are the steps to reproduce this issue?
1. Make a simple python script that blinks 10 times (no loops, just copy paste led.toggle() & sleep(0.25) - name it main.py
2. Upload current file to Pico
3. Hard reset

## What happens?
If led is toggled every 0,25 sec - it will turn on once and turns off and **stops execution**
If led is toggled every 1 sec it will turn on and **stops execution**
So it's either time based or based on **how fast the board connects to the laptop** via usb

## What were you expecting to happen?
Board restarts and script executes every line of code

## Any other comments?
Replace step 3 with "Soft reset & listen" and script **will work in it's entirety**. Same happens if I unplug/plug the board when VScode is closed (I guess when vREPL is not running) - it works. It also works fine when run manually (via Run button)

**tl;dr** - autorun main.py script stops almost immediately when the board is plugged via USB to the computer that is running MicroPico in VSCode. It runs fine when plugged to a laptop that doesn't run MicroPico, or plugged into a power brick. I've tested this on three different boards, one brand new - same behaviour

## Which version of MicroPico are you using?
v3.4.1

## Support info
```
Version: 1.83.1 (user setup)
Commit: f1b07bd25dfad64b0167beb15359ae573aecd2cc
Date: 2023-10-10T23:48:05.904Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.19045
```

lefixx, 2023-10-22:
When you connect the board to the pc running vscode the pc stops execution to connect to the board. Try closing vscode or powering the board from a different power source

WhatNot911, 2023-10-22:
> When you connect the board to the pc running vscode the pc stops execution to connect to the board. Try closing vscode or powering the board from a different power source

Yeah, I mentioned this in "Any other comments?" section. Doesn't change anything much, should the board be able to run code regardless?

lefixx, 2023-10-23:
Sounds to me as intended behavior. When you connect to the board execution stops so you can connect to it. There is a setting to turn this of (IIRC). It is also easy to work around (by connecting and then running main.py manually)

lefixx, 2023-10-23:
may be relevant: https://github.com/paulober/MicroPico/issues/92

paulober, 2023-10-30:
@WhatNot911 Take a look at [listen2pico](https://github.com/paulober/listen2pico) utility for now. It allows you to listen to the MicroPython output without interrupting running programs.

paulober, 2023-10-30:
Like @lefixx mentioned, it's a duplicate of #92.

webdeb, 2023-12-26:
@paulober the problem with this is, that when you upload a change of a file and the board restarts, then the `main.py` starts automatically which in my case is the initialization of `asyncio`.. After the board has started, the extension connects to the board and then all running programs stops. This is very inconvenient if you are building smth. bigger.

What I am doing now, is: after any upload, I've to disconnect the PICO and reset it. I am planning to give the board to customers and let them write plugins, the process of developing the plugins for it, would be very straightforward with MicroPico, if this little problem would not be there.

I think it would be sufficient to provide a custom `dev.py` script, or `micropico.py` which just does an `import main` or whatever to not make the board stuck, when micropico connects to it.

Please check my main.py (its basically the recommended asyncio application)

```py
import uasyncio as asyncio
from lib.user_inputs import create_inputs
from main_menu import create_main_menu

def set_global_exception():
    def handle_exception(loop, context):
        import sys
        sys.print_exception(context["exception"])
        sys.exit()
    loop = asyncio.get_event_loop()
    loop.set_exception_handler(handle_exception)

async def start_app():
    await create_inputs()
    await create_main_menu()

async def main():
    set_global_exception()
    loop = asyncio.get_event_loop()
    loop.create_task(start_app())
    loop.run_forever()

def start():
    print("Starting...")
    try:
        asyncio.run(main())
    except:
        print("Started new event loop")
        asyncio.new_event_loop()

start()
```

paulober, 2023-12-26:
Ok, so you would like following approach the extension to connect, reset and then run a custom entry point (if present) like `micropico.py` (`mp.py`) or `dev.py`?

webdeb, 2023-12-26:
@paulober yeah, I think this would be a sufficient workaround for most cases, because it is then on the side of the developer to decide how to go from that point on, restore previous state, or restart. One could enable debug prints for example..

webdeb, 2023-12-26:
The script itself could be just a file inside the project directory, don't even need to be on the board, if just used for running once?

paulober, 2023-12-26:
> The script itself could be just a file inside the project directory, don't even need to be on the board, if just used for running once?

I could also only add a setting for your vscode settings.json where you could set a filename on the board to be executed if present on connection with MicroPico. Does this sound good?

webdeb, 2023-12-26:
> I could also only add a setting for your vscode settings.json where you could set a filename on the board to be executed if present on connection with MicroPico. Does this sound good?

@paulober sounds perfect!

webdeb, 2023-12-27:
One question, would the file be executed also if the file is inside the firmware?

paulober, 2023-12-27:
> One question, would the file be executed also if the file is inside the firmware?

It depends on the file being visible in the filesystem. Though I could also add a setting that auto-imports a module like that's included in the firmware.

webdeb, 2023-12-27:
I think that would be more generic 👍  would fit for all module fs and firmware stuff. Thank you so much, I'll test it asap.

commit 61391b28 (2023-12-27): Fix #153 #176, Added executeOnConnect setting
closed 2023-12-27 by paulober