paulober/MicroPico #154: Package management/install

tompinnacle, 2023-10-21:
Maybe I'm missing something obvious, but how do you acquire and add a package when using VS Code and MicroPico?

In Thonny there is a package manager
When using full Python you can use pip

But I haven't found anything about doing it in VS Code

paulober, 2023-10-24:
In MicroPython there is no pip available and of course pip installs packages locally. 
My understanding is that Thonny's package manager UI is only a GUI interface for pip on your local machine.

Like @lefixx mentioned, you can either include a library if it's a single file into your project or have a look at: https://docs.micropython.org/en/latest/reference/packages.html
I may work on a GUI for installing mip packages via VSCode on your pico but that my take a while...

lefixx, 2023-10-25:
I will take a look at mip, thanks.
So installing libraries to the project and then uploading the whole project is not a thing?

paulober, 2023-10-25:
No, cause the files are install in your pip directory. Though you can install it with `pip install --target <path/to/folder/in/project> <package-name>` into your project directory and then upload it with the upload-project command.

lefixx, 2023-10-25:
That sounds as an even better solution. many thanks

lark-cig, 2023-10-31:
I have started to use pipkin, it can upload directly to the board, but I download to my Mac and then upload to pico from 
VS Code. The Micropico documentation mentions a sync-button, but I haven't found one so I upload via the popup menu.

for example: pipkin -d ./lib install micropython-oled