paulober/MicroPico #155: Working with two PICOs

PavelKalina, 2023-10-23:
Switch Pico command switch only to higher port number.

## What are the steps to reproduce this issue?
1. Tried to create both two independent projects and one project.
2. autoConnect: false
3. manualComDevice: empty or lower COM number.

## What happens?
switchPico command works only to switch to higher COM.

## What were you expecting to happen?
To switch continuously to COM with lower number.

## Any logs, error output, etc?
repetitive message "Disconnected from Pico"

## Any other comments?
I would like to work with two PICOs in parallel (Tx, Rx)

## Which version of MicroPico are you using?
3.4.1 (newer worked)

## Support info

```
Version: 1.83.1 (user setup)
Commit: f1b07bd25dfad64b0167beb15359ae573aecd2cc
Date: 2023-10-10T23:48:05.904Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.19044
```

paulober, 2023-10-24:
Thanks for reporting, I'll look into it...

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around connecting and reconnecting boards. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.

closed 2026-10-22 by github-actions[bot]