paulober/MicroPico #156: Debugging / ModuleNotFoundError

HeinMuel, 2023-10-23:
I have all the extension installed and the toggle-led project created.
All works fine if i run the project on the pi pico.
If I try to debug the project with VSC the error above is showed.
In the project folder there are all of the module e.g. ".vscode/Pico-W-Stub/machine.pyi".
Debugging works if I place the machine.pyi beside the test.py file, but I have to rename it to .py.
Where can I tell the debugger how to find the modules?

Thank you for your help!

I work on the newest version of linux mint.
If I try to make the project on a SMB-network-server,  "configure network " fails, no Pico-W-Stub folder is created.

paulober, 2023-10-24:
Debugging in VSCode would try to run the script on your computer, so it would not make sense to make the stubs available to the debug as they do not contain source-code only the stubs. But I'll look into debug options...

Working on an SMB network share, doesn't work because it's not possible to create a directory junction between the stubs folder on your local drive and a folder in a network share.

HeinMuel, 2023-10-25:
Thank you for the answer!
Please tell me if you discover a buck.
Perhaps it is only a wrong setting so the debugger don´t take the Pico to run the program.
But I can make a normal run, LED is blinking on the board and I get the right output in the terminal.
Best regards!

paulober, 2023-10-25:
It's not a bug, debugging does not work, because the normal python debug in VSCode does run the python scripts locally. And yes, debugging does not throw when playing machine.pyi next to your script, because it the can import it. But machine.pyi doesn't do anything on your machine so it has no use.

HeinMuel, 2023-10-26:
Thank you for your help, although I am sad because I cannot debug.

MarByteBeep, 2023-12-05:
> Thank you for your help, although I am sad because I cannot debug.

It's not ideal, but you can always try to debug using `print` statements everywhere.

paulober, 2026-09-14:
Thanks for the report! This is the same problem as #203, so I'm closing this one to keep everything in one place.

closed 2026-09-14 by paulober