paulober/MicroPico #157: Could not run program by the error "Exception occured (maybe a connection loss)" on VS Code. (I need help)

Actonmy, 2023-10-25:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Installed Micro-Pico extension to VS Code.
2. Confirmed connection Raspberry Pi Pico and PC by USB (Com8 is not conflicted other ports.)
3. Use flash LED program and Run current file on Pico. 

## What happens?
Could not run by the error "Exception occured (maybe a connection loss)"

## What were you expecting to happen?

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?
Micro-Pico c
Python v2023.18.0
Raspberry Pi Pico WH MicroPython firmware  RPI_PICO_W-20231005-v1.21.0.uf2
Visual Studio Code v

## Which version of MicroPico are you using?
3.4.1

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.83.1 (user setup)
Commit: f1b07bd25dfad64b0167beb15359ae573aecd2cc
Date: 2023-10-10T23:48:05.904Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.19045
```

PavelKalina, 2023-10-26:
Try in Thonny to clarify the problem.

Actonmy, 2023-10-27:
Program works correctly in Thonny 4.1.2.

paulober, 2023-10-29:
@Actonmy 
Do you have any output in `Output > MicroPico`?
Is it possible to look at the script you're encountering the problem with?

Actonmy, 2023-10-30:
Hello Paulober,  Thank you. The logs are as below.

2023-10-30 13:25:30.231 [info] Extension host with pid 15252 started
2023-10-30 13:25:30.231 [info] Skipping acquiring lock for c:\Users\user\AppData\Roaming\Code\User\workspaceStorage\a5e733c2bec36c88d1b2c7cc082abe58.
2023-10-30 13:25:31.215 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-10-30 13:25:31.219 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python', root cause: ms-python.isort
2023-10-30 13:25:31.223 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter, startup: false, activationEvent: 'onLanguage:python'
2023-10-30 13:25:31.226 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2023-10-30 13:25:31.915 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2023-10-30 13:25:33.409 [warning] Using default string since no string found in i18n bundle that has the key: You need to select a Python interpreter before you start debugging.

Tip: click on "Select Interpreter" in the status bar.
2023-10-30 13:25:33.409 [warning] Using default string since no string found in i18n bundle that has the key: The Python extension now includes Pylance to improve completions, code navigation, overall performance and much more! You can learn more about the update and learn how to change your language server [here](https://aka.ms/new-python-bundle).

Read Pylance's license [here](https://marketplace.visualstudio.com/items/ms-python.vscode-pylance/license).
2023-10-30 13:25:33.414 [warning] Using default string since no string found in i18n bundle that has the key: The Microsoft Python Language Server has reached end of life. Your language server has been set to the default for Python in VS Code, Pylance.

If you'd like to change your language server, you can learn about how to do so [here](https://devblogs.microsoft.com/python/python-in-visual-studio-code-may-2021-release/#configuring-your-language-server).

Read Pylance's license [here](https://marketplace.visualstudio.com/items/ms-python.vscode-pylance/license).
2023-10-30 13:25:34.583 [warning] Using default string since no string found in i18n bundle that has the key: Select 'Existing JupyterHub Server...' from the Kernel Picker
2023-10-30 13:25:34.584 [warning] Using default string since no string found in i18n bundle that has the key: Use this option when connecting to JupyterHub servers
2023-10-30 13:25:34.584 [warning] Using default string since no string found in i18n bundle that has the key: Once installed, open the Kernel Picker and select 'Existing JupyterHub Server...'
2023-10-30 13:25:34.584 [warning] Using default string since no string found in i18n bundle that has the key: Python is not installed. 
Please download and install Python in order to execute cells in this notebook. 
Once installed please reload VS Code.
2023-10-30 13:25:34.586 [warning] Using default string since no string found in i18n bundle that has the key: Please check the 'Jupyter' [output](command:jupyter.viewOutput) panel for further details./Do not translate the text "command:jupyter.viewOutput", that is a command Id that will be used by VS Code to open the output panel
2023-10-30 13:25:34.588 [warning] Using default string since no string found in i18n bundle that has the key: Jupyter Server Console
2023-10-30 13:25:34.589 [warning] Using default string since no string found in i18n bundle that has the key: Operation not supported in web version of Jupyter Extension.
2023-10-30 13:25:34.590 [warning] Using default string since no string found in i18n bundle that has the key: Failed to install the Python Extension.
2023-10-30 13:25:40.177 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-10-30 13:25:40.199 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.micropico'
2023-10-30 13:25:42.545 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter-renderers, startup: false, activationEvent: 'api', root cause: ms-toolsai.jupyter
2023-10-30 13:25:43.282 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-10-30 13:25:43.285 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-10-30 13:25:43.292 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl-recommender, startup: false, activationEvent: 'onStartupFinished'
2023-10-30 13:25:43.296 [info] ExtensionService#_doActivateExtension vscode-icons-team.vscode-icons, startup: false, activationEvent: 'onStartupFinished'
2023-10-30 13:25:44.078 [info] ExtensionService#_doActivateExtension ms-python.isort, startup: false, activationEvent: 'onLanguage:python'
2023-10-30 13:25:44.082 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-10-30 13:25:46.445 [error] No bundle location found for extension ms-vscode-remote.remote-wsl-recommender
2023-10-30 13:25:47.636 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-10-30 13:25:47.639 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-10-30 13:26:00.662 [info] Eager extensions activated

paulober, 2023-10-30:
Thanks but I meant `Output pane -> Select 'MicroPico' in the dropdown`.

Actonmy, 2023-10-31:
Sorry, I missunderstood. Yes, push Ctrl + Shift + p keys and Micropico: configure project was selected. Then I run the script by "run current file on pico" in dropdown,  Terminal (Pico vREPL) shows Exception occured (maybe a connection loss).
The script was used example of LED flash.

rodolfoBocanegra, 2023-11-27:
@Actonmy did you find a solution on this? I have the same problem

paulober, 2023-11-27:
@rodolfoBocanegra are you also on Windows x64?

@Actonmy I'll make you both a debug version of the extension which should give us more feedback about what goes wrong there.

paulober, 2023-11-27:
@Actonmy And sorry for the delay

rodolfoBocanegra, 2023-11-27:
@paulober yes, I'm on windows 11, x64 architecture, running micropython on a raspberry pi pico H, I do have the extensions required in the landing page (vs code extension) of micropico, I've tried running VS Code as admin, rebooting rp2040 device, executing script on thonny (which works normal) and nothing seems to change the output; and thank you for responding!

jonascinarkiziltepe, 2023-12-05:
I have the same problem, I also tried to upload the program file as "main.py" using thonny and it works as intended, but when I try the download project command from micropico it loses connection. Also I can run code on pico using the micropico extension with right click > "Run current selection on pico" command

paulober, 2023-12-06:
@jonascinarkiziltepe and @rodolfoBocanegra here is the debug build I promised. Please reproduce the error and send me the output in the output panel > MicroPico (in the dropdown).
OneDrive: https://1drv.ms/u/s!AvaW5dhFqJnIh_hp8ZzfQwpdE3IYRA?e=e4A8qI

You can install the custom build in the extensions panel, three dots top right hand corner "install from vsix" (to make sure it works, please uninstall the current build from the store)

rodolfoBocanegra, 2023-12-07:
Thank you @paulober ! I followed your instructions and these are my logs after trying to run my program:

```
2023-12-07 02:13:08.815 [info] [Stubs] Installed stubs are already up to date!
2023-12-07 02:13:12.210 [info] [Activator] Connection to wrapper successfully established
2023-12-07 02:13:12.965 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:13.015 [warning] [PicoWFs] stat: item not found: /.devcontainer/devcontainer.json
2023-12-07 02:13:13.111 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:13.226 [warning] [PicoWFs] stat: item not found: /pom.xml
2023-12-07 02:13:13.312 [warning] [PicoWFs] stat: item not found: /node_modules
2023-12-07 02:13:13.907 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:13.957 [warning] [PicoWFs] stat: item not found: /build.gradle
2023-12-07 02:13:13.998 [warning] [PicoWFs] stat: item not found: /pom.xml
2023-12-07 02:13:14.204 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:14.272 [warning] [PicoWFs] stat: item not found: /pyproject.toml
2023-12-07 02:13:14.314 [warning] [PicoWFs] stat: item not found: /app/src/main/AndroidManifest.xml
2023-12-07 02:13:14.328 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:14.371 [warning] [PicoWFs] stat: item not found: /.pep8
2023-12-07 02:13:14.437 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:14.478 [warning] [PicoWFs] stat: item not found: /setup.cfg
2023-12-07 02:13:14.492 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:14.533 [warning] [PicoWFs] stat: item not found: /tox.ini
2023-12-07 02:13:14.563 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:14.604 [warning] [PicoWFs] stat: item not found: /.pylintrc
2023-12-07 02:13:14.634 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:14.676 [warning] [PicoWFs] stat: item not found: /pylintrc
2023-12-07 02:13:14.734 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:14.776 [warning] [PicoWFs] stat: item not found: /.mypy.ini
2023-12-07 02:13:14.810 [error] [PicoWFs] readFile: File not found
2023-12-07 02:13:14.851 [warning] [PicoWFs] stat: item not found: /.flake8
2023-12-07 02:13:43.790 [info] [Activator] Pyboard exited with code 0
2023-12-07 02:13:45.009 [info] [Activator] Connection to wrapper successfully established
```
My project does have one `.micropico` file containing:
```
{
    "info": "This file is just used to identify a project folder."
}
```

paulober, 2023-12-07:
@rodolfoBocanegra Thanks. Did you still get the error with connection loss?
Cause it should also now be printed in the output panel... In case you did, here is a new version of the debug build, that should print the connection-loss error in `output > MicroPico` or `output > Extension Host`:
https://1drv.ms/u/s!AvaW5dhFqJnIh_htc9ARejv7MxV2lQ?e=jgkcWy

Please make sure to uninstall the current version before installing the debug build. And thanks for testing.

rodolfoBocanegra, 2023-12-07:
@paulober thank you for providing this new build! 

After removing the last build you provided (called `pico-w-go-3.5.0-1.vsix`) and installed through VSIX the one downloaded from: https://1drv.ms/u/s!AvaW5dhFqJnIh_htc9ARejv7MxV2lQ?e=jgkcWy (called `pico-w-go-3.5.0.vsix`) now I'm not able to run my file in my pico, if I try to configure my project, run it or to connect or to disconnect the pico (all this through `ctrl `+ `shift `+ `P`) I only get one message at the bottom of my VS code saying 'Activating Extensions ...' which lasts for around 5 secs and then disapears but nothing happens, I even lost my last options in the status bar like "Pico connected", "Run", "Reset", etc (which should appear right where 'Activating Extensions ...' sign is).

This is how my status bar looks like:
[image]

Is this the right build? I'd like to point out the last build you sent is about 18 MB while the 1st debug file is around 61MB:

| File name | Size | URL |
|----------|----------|----------|
| pico-w-go-3.5.0-1.vsix    | 61,034 KB   | https://1drv.ms/u/s!AvaW5dhFqJnIh_hp8ZzfQwpdE3IYRA?e=e4A8qI   |
| pico-w-go-3.5.0.vsix    | 18,063 KB  | https://1drv.ms/u/s!AvaW5dhFqJnIh_htc9ARejv7MxV2lQ?e=jgkcWy   |

> @rodolfoBocanegra Thanks. Did you still get the error with connection loss? Cause it should also now be printed in the output panel... In case you did, here is a new version of the debug build, that [...]

rodolfoBocanegra, 2023-12-07:
@paulober thank you!

Steps I just performed:
1. Uninstalled last (15 MB build) 
2. Installed new latest provided build
3. Restarted VS code
4. Tried to run my script on my pico
5. Same `Exception occured (maybe a connection loss)` message from terminal

Logs from `Output`->`MicroPico`:
```
2023-12-07 14:56:26.791 [info] [Stubs] Installed stubs are already up to date!
2023-12-07 14:56:27.282 [info] [Activator] Connection to wrapper successfully established
2023-12-07 14:56:30.232 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:30.274 [warning] [PicoWFs] stat: item not found: /.devcontainer/devcontainer.json
2023-12-07 14:56:30.438 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:30.501 [warning] [PicoWFs] stat: item not found: /pom.xml
2023-12-07 14:56:30.549 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:30.591 [warning] [PicoWFs] stat: item not found: /build.gradle
2023-12-07 14:56:30.648 [warning] [PicoWFs] stat: item not found: /node_modules
2023-12-07 14:56:30.659 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:30.701 [warning] [PicoWFs] stat: item not found: /pyproject.toml
2023-12-07 14:56:30.820 [warning] [PicoWFs] stat: item not found: /pom.xml
2023-12-07 14:56:30.830 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:30.872 [warning] [PicoWFs] stat: item not found: /.pep8
2023-12-07 14:56:31.094 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:31.163 [warning] [PicoWFs] stat: item not found: /setup.cfg
2023-12-07 14:56:31.348 [warning] [PicoWFs] stat: item not found: /app/src/main/AndroidManifest.xml
2023-12-07 14:56:31.406 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:31.448 [warning] [PicoWFs] stat: item not found: /tox.ini
2023-12-07 14:56:31.462 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:31.503 [warning] [PicoWFs] stat: item not found: /.pylintrc
2023-12-07 14:56:31.515 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:31.555 [warning] [PicoWFs] stat: item not found: /pylintrc
2023-12-07 14:56:31.598 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:31.638 [warning] [PicoWFs] stat: item not found: /.mypy.ini
2023-12-07 14:56:31.660 [error] [PicoWFs] readFile: File not found
2023-12-07 14:56:31.700 [warning] [PicoWFs] stat: item not found: /.flake8
2023-12-07 14:56:46.310 [error] [Activator] Exception occured (maybe a connection loss). Message dump: !!ERR!!
2023-12-07 14:56:46.323 [info] [Activator] Pyboard exited with code 0
2023-12-07 14:56:47.635 [info] [Activator] Connection to wrapper successfully established

```

Logs from `Output`->`Extension host`
```
2023-12-07 14:56:24.752 [info] Extension host with pid 1448 started
2023-12-07 14:56:24.752 [info] Skipping acquiring lock for c:\Users\rodol\AppData\Roaming\Code\User\workspaceStorage\6b8c39270ac562cadfcc791ce39aaac1.
2023-12-07 14:56:25.019 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onFileSystem:pico'
2023-12-07 14:56:25.045 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-12-07 14:56:25.053 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python', root cause: ms-python.black-formatter
2023-12-07 14:56:25.199 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2023-12-07 14:56:25.369 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2023-12-07 14:56:26.004 [info] ExtensionService#_doActivateExtension ms-python.black-formatter, startup: false, activationEvent: 'onLanguage:python'
2023-12-07 14:56:26.027 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-12-07 14:56:26.353 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-12-07 14:56:26.544 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-12-07 14:56:26.600 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-12-07 14:56:28.994 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-12-07 14:56:29.254 [info] Eager extensions activated
2023-12-07 14:56:29.340 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-12-07 14:56:29.347 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-12-07 14:56:29.356 [info] ExtensionService#_doActivateExtension dbaeumer.vscode-eslint, startup: false, activationEvent: 'onStartupFinished'
2023-12-07 14:56:29.369 [info] ExtensionService#_doActivateExtension eamodio.gitlens, startup: false, activationEvent: 'onStartupFinished'
2023-12-07 14:56:29.536 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-containers, startup: false, activationEvent: 'onStartupFinished'
2023-12-07 14:56:29.701 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl, startup: false, activationEvent: 'onStartupFinished'
2023-12-07 14:56:30.353 [error] [ms-python.python] provider FAILED
[... cut ...]
2023-12-07 14:56:43.196 [error] Error: jedi.api.environment.InvalidPythonEnvironment: Scripts\python.exe seems to be missing.
	at c:\Users\rodol\.vscode\extensions\ms-python.python-2023.22.0\out\client\extension.js:2:2338987
	at re (c:\Users\rodol\.vscode\extensions\ms-python.python-2023.22.0\out\client\extension.js:2:2339281)
	at c:\Users\rodol\.vscode\extensions\ms-python.python-2023.22.0\out\client\extension.js:2:2334065
	at Immediate.<anonymous> (c:\Users\rodol\.vscode\extensions\ms-python.python-2023.22.0\out\client\extension.js:2:2334085)
	at process.processImmediate (node:internal/timers:476:21)
```

paulober, 2023-12-07:
Ok, I'd expect a different output but I'll look into it. Could It be that the storage on your pico is full or that the folder you're trying to upload is to big for the picos flash?

paulober, 2023-12-07:
Strange, and what happens if you upload just a folder with one file in it with the upload project function?

paulober, 2023-12-07:
Thanks for all the testing. I'll investigate possible problem sources that can cause this problem.

rodolfoBocanegra, 2023-12-07:
Thank you! Just to confirm, besides having the vs code recommended extensions and the micropython firmware installed in my pico, is there any other requirement my machine has to comply with?

paulober, 2023-12-07:
Maybe trying a different USB port and/or cable if it's some strange usb stuff that's causing this connection issue...
But if it's working in a different program I guess that's not the issue.
And if you're able to connect to the pico and run a command on it all requirements are met.

paulober, 2023-12-07:
@rodolfoBocanegra I might found a different solution to see where you got the error. Are you available for testing another debug build?

paulober, 2023-12-07:
Great, here is a new build. You are on Windows x64, right?: https://1drv.ms/u/s!AvaW5dhFqJnIh_kOr1T24i4FSeiNPA?e=kKzMhf

Again uninstalling the old one and then vsix. The logs should now contain the reason why it fails.

paulober, 2023-12-07:
I guess Windows defender is blocking the wrapper since it's a new unknown build. Can you somehow whitelist it?

paulober, 2023-12-16:
> My apologies, I'm going to be busy for a some hours, will get back with updates soon

Any updates on this?

rodolfoBocanegra, 2023-12-17:
My apologies for this late response, after reinstalling through VSIX 3 times the last dev build you provided (all this with and without anti virus protection) I haven't been able to make it work (options/buttons are never added to status bar) are you sure build provided at: https://1drv.ms/u/s!AvaW5dhFqJnIh_kOr1T24i4FSeiNPA?e=kKzMhf is a valid build?

However I've found a couple of curious behaviors with the current online version provided at VS code store, example:

> Note: please remember these next points reffer to MicroPico current online version, NOT dev build.

1. When pressing `toggle Pico-W-FS` button, the pico content succesfully appears in my VS Code explorer as shown bellow (red rectangle):
[image]
2. After experimenting a bit:
  - I'm able to create, open, modify and delete files in my pico which ensures data IO works.
  - I'm able to run files in my pico from this latest explorer by: `right click` -> `Remote > run current file on Pico` as shown bellow:
  
[image]

  - I'm NOT able to upload projects to my pico by: `Ctrl` + `Shift` + `P` -> `MicroPico: Upload project to Pico` (even after toggling pico FS)
  - When trying to upload a project or a file to my pico next happens:
    - `Connection to pico lost` notification appears.
    - Another `Uploading file` or `Uploading project` notification appears and runs forever as shown bellow:

[image]

2. I've tested too this experiments with another Pico which storage is empty and all behaviors are the same.

I hope this comments adds information to this issue and sorry for the inconvenience and the delay.

Thank you!

NB928, 2024-01-30:
Any updates? 
I can run via terminal (REPL) instead but I can't stop with ctrl-c so a bit inconvenient... 
After Run I get "Disconnected from Pico" message, Then it connects again.

2024-01-29 12:59:57.456 [info] [Activator] Pyboard exited with code 0
2024-01-29 12:59:57.988 [info] [Activator] Connection to wrapper successfully established

Problem is on my windows (10). On macOS it works.

paulober, 2024-01-30:
@rodolfoBocanegra Thanks for this detailed report on your observations. I guess the issue might be related to a buffer issue. I'll reach back if I find something out.

paulober, 2024-01-30:
@NB928 Ok, so you don't have the same issue as @rodolfoBocanegra but ctrl-c doesn't work on Windows but on macOS?

NB928, 2024-01-31:
I have the same issue as @rodolfoBocanegra. I guess he can also run via REPL. The ctrl-c does not work on macOS either but there I have no problem since your Run/Stop button works as intended.

I can view what files are on my pico with Pico-W-FS but I can't open a file from the list. It disconnects and waits forever.

NB928, 2024-02-01:
Maybe this in terminal.mts prevents ctrl-c to be sent if something is running:

      } else if (char === "\x03") {
        // Ctrl+C
        if (!this.waitingForPrompt) {
          return;
        }

        this.submitEmitter.fire(char);

paulober, 2024-02-02:
@rodolfoBocanegra To find the issue faster I would like to ask you if you can run some commands for me on the machine that has this issue?

```bash
cd ~/.vscode/extensions/paulober.pico-w-go-3.7.4-win32-x64/dist/scripts/wrapper_win32
wrapper_win32_amd64.exe --scan-ports
wrapper_win32_amd64.exe -d <the port shown before> -b 115200
```

and then insert following json and press enter (one by one):

- please replace /path/to/myupload.py with a path to a test python file to upload
- you should get similar output like shown below except for the upload command

```json
{ "command": "status" }
!!EOO!!
{ "command": "command", "args": { "command": "print('HI')\n" } }
HI
!!EOO!!
{ "command": "list_contents_recursive", "args": { "target": "/" } }
          95 /boot.py
         169 /main.py
!!EOO!!
{ "command": "upload_files", "args": { "files": ["/path/to/myupload.py"], "remote": "/" } }
!!EOO!!
{ "command": "list_contents_recursive", "args": { "target": "/" } }
          95 /boot.py
         137 /myupload.py
         169 /main.py
!!EOO!!
{ "command": "exit" }
```

NB928, 2024-02-05:
C:\Users\ÄGARE\.vscode\extensions\paulober.pico-w-go-3.7.4-win32-x64\dist\scripts\wrapper_win32>wrapper_win32_amd64.exe --scan-ports
COM4,115200
!!EOO!!

C:\Users\ÄGARE\.vscode\extensions\paulober.pico-w-go-3.7.4-win32-x64\dist\scripts\wrapper_win32>wrapper_win32_amd64.exe -d COM4 -b 115200

here it never returns.................

NB928, 2024-02-05:
Hi, I ran the above and got the same as you. The file was uploaded without any errors

paulober, 2024-02-05:
@NB928 Thanks for all the testing. Here is a debug build which opens a new output channel in vscode called "MpyDebug". Please uninstall the current version of the extension the select the three dots in the extensions panel and select install from vsix and install the vsix file. If you then reproduce the issue you should get extensive debug output in the "MpyDebug" channel: https://1drv.ms/u/s!AvaW5dhFqJnIh_1PwUYkqJYRn0pvng?e=PN2MeS (again OneDrive because GitHub doesn't allow uploads this size)

NB928, 2024-02-08:
Could it be the character "Ä" that is the problem?

2024-02-08 18:58:02.690 [info] Command: {"command":"status","args":{}}
2024-02-08 18:58:02.693 [info] !!EOO!!

2024-02-08 18:58:03.853 [info] Command: {"command":"command","args":{"command":"\rfrom usys import implementation, version; print(version.split('; ')[1] + '; ' + implementation._machine)"}}
2024-02-08 18:58:03.859 [info] M
2024-02-08 18:58:03.859 [info] icroPy
2024-02-08 18:58:03.859 [info] thon 
2024-02-08 18:58:03.859 [info] v1.22.
2024-02-08 18:58:03.860 [info] 1 on 
2024-02-08 18:58:03.860 [info] 2024-
2024-02-08 18:58:03.860 [info] 01-
2024-02-08 18:58:03.860 [info] 05; W
2024-02-08 18:58:03.860 [info] 5500
2024-02-08 18:58:03.860 [info] -EVB
2024-02-08 18:58:03.860 [info] -Pico
2024-02-08 18:58:03.861 [info]  wit
2024-02-08 18:58:03.861 [info] h R
2024-02-08 18:58:03.861 [info] P2
2024-02-08 18:58:03.861 [info] 04
2024-02-08 18:58:03.861 [info] 0

2024-02-08 18:58:03.861 [info] !!EOO!!

2024-02-08 18:58:03.946 [info] Command: {"command":"run_file","args":{"files":["c:\\Users\\ÄGARE\\Documents\\VSCode\\PL800\\Controller\\test.py"]}}
2024-02-08 18:58:03.946 [info] !!ERR!!
!!EOO!!

2024-02-08 18:58:07.695 [info] Command: {"command":"status","args":{}}
2024-02-08 18:58:07.700 [info] !!EOO!!

NB928, 2024-02-08:
upploading single file:

2024-02-08 19:01:03.913 [info] Command: {"command":"command","args":{"command":"import gc as __pico_gc; __pico_gc.collect(); del __pico_gc"}}
2024-02-08 19:01:03.918 [info] !!EOO!!

2024-02-08 19:01:03.919 [info] Command: {"command":"upload_files","args":{"files":["c:\\Users\\ÄGARE\\Documents\\VSCode\\PL800\\Controller\\test.py"],"remote":"/","verbose":true}}
2024-02-08 19:01:03.920 [info] !!Exception!!
[WinError 3] The system cannot find the path specified: 'c:/Users/ÄGARE/Documents/VSCode/PL800/Controller/test.py'

2024-02-08 19:01:08.231 [info] Command: {"command":"status","args":{}}
2024-02-08 19:01:08.235 [info] !!EOO!!

Overlord6644, 2024-02-24:
Hello currently having the same issue is there any method to solve it or work around ?

paulober, 2026-09-14:
This was reported against MicroPico 3.x, which used a connection backend that version 4 replaced completely. If you still see this on the latest version, please open a new issue. Thanks!

closed 2026-09-14 by paulober