paulober/MicroPico #159: Micropico stopped working after version 3.2.6

zopp99, 2023-11-02:
MicroPico versions 3.3.0 through 3.4.1 do not appear to do anything when MicroPico commands are executed.  (Downgrading to MicroPico version 3.2.6 works fine).
  
Beginning at version 3.3.0, I no longer see Micropico 'buttons' on the bottom status bar.  Executing commands directly via the command pallet briefly shows a 'activating extensions' message but nothing happens.  There are no messages in the MircroPico log except for one initial line showing 'Installed stubs are already up to date. 

## What are the steps to reproduce this issue?
1.  Install MicroPico 3.4.1
2.  Navigate to any prior valid micropico folder / upython program
3.  Execute any micropico command.    -> nothing appears to happen

## What happens?
No MicroPico 'buttons' on the bottom vscode status bar.  When executing commands directly via command pallet, nothing appears to happen.  No log messages are generated. 

## What were you expecting to happen?
MicroPico buttons should be displayed at bottom of vscode status bar.  MicroPico commands should be executed.

## Any logs, error output, etc?
MicroPico Output log consists of one line:
2023-11-02 09:38:02.916 [info] [Stubs] Installed stubs are already up to date!

## Any other comments?
Thank you for all your efforts on this excellent product! 

## Which version of MicroPico are you using?
3.4.1

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.84.0
Commit: d037ac076cee195194f93ce6fe2bdfe2969cc82d
Date: 2023-11-01T11:28:26.881Z
Electron: 25.9.2
ElectronBuildId: 24603566
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux arm64 6.1.21-v8+
```

paulober, 2023-11-03:
Any output in `Output > Extension Host`?

zopp99, 2023-11-03:
yes sir.  Output > Extension Host as follows.  Thank you
2023-11-02 09:37:44.855 [info] Extension host with pid 40250 started
2023-11-02 09:37:44.856 [info] Skipping acquiring lock for /home/pi/.config/Code/User/workspaceStorage/1b4aada35475271171d69089ffb6b4c6.
2023-11-02 09:37:46.047 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-11-02 09:37:46.178 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python', root cause: ms-python.isort
2023-11-02 09:37:46.711 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter, startup: false, activationEvent: 'onLanguage:python'
2023-11-02 09:37:47.474 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2023-11-02 09:37:47.965 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2023-11-02 09:37:51.814 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-11-02 09:37:51.846 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.micropico'
2023-11-02 09:37:52.967 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter-renderers, startup: false, activationEvent: 'api', root cause: ms-toolsai.jupyter
2023-11-02 09:37:55.614 [info] Not activating extension 'ms-vscode.cmake-tools': Timed out while searching for 'workspaceContains' pattern */CMakeLists.txt,*/*/CMakeLists.txt,*/*/*/CMakeLists.txt
2023-11-02 09:37:56.354 [info] ExtensionService#_doActivateExtension ms-python.isort, startup: false, activationEvent: 'onLanguage:python'
2023-11-02 09:37:56.584 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-11-02 09:37:59.412 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-11-02 09:37:59.511 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-11-02 09:38:00.391 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-11-02 09:38:00.402 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-11-02 09:38:00.451 [info] ExtensionService#_doActivateExtension EthanSK.restore-terminals, startup: false, activationEvent: 'onStartupFinished'
2023-11-02 09:38:00.499 [info] ExtensionService#_doActivateExtension mcu-debug.debug-tracker-vscode, startup: false, activationEvent: 'onStartupFinished', root cause: marus25.cortex-debug
2023-11-02 09:38:00.512 [info] ExtensionService#_doActivateExtension mcu-debug.peripheral-viewer, startup: false, activationEvent: 'onStartupFinished', root cause: marus25.cortex-debug
2023-11-02 09:38:00.589 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-containers, startup: false, activationEvent: 'onStartupFinished'
2023-11-02 09:38:00.945 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl, startup: false, activationEvent: 'onStartupFinished'
2023-11-02 09:38:02.304 [info] ExtensionService#_doActivateExtension mcu-debug.memory-view, startup: false, activationEvent: 'onStartupFinished', root cause: marus25.cortex-debug
2023-11-02 09:38:02.513 [info] ExtensionService#_doActivateExtension mcu-debug.rtos-views, startup: false, activationEvent: 'onStartupFinished', root cause: marus25.cortex-debug
2023-11-02 09:38:04.439 [info] ExtensionService#_doActivateExtension marus25.cortex-debug, startup: false, activationEvent: 'onStartupFinished'
2023-11-02 09:38:13.443 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-11-02 09:38:15.521 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-11-02 09:38:15.681 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'

paulober, 2023-11-03:
Thanks, nothing unusual there. If you right-click the status-bar at the bottom, do you see MicroPico in the context menu?

zopp99, 2023-11-04:
No, there is no 'MicroPico (Extension)' listed in the context menu of the bottom status bar for version 3.4.1,   (For my reference, I reinstalled version 3.2.6 and  'MicroPico (Extension)' is listed in the context menu in the bottom status bar.  But to restate, it is not listed with version 3.4.1)   Thanks again for your help.

zopp99, 2023-11-04:
I should add....  Interestingly, while MicroPico is not in the bottom status bar, It is in the Command Pallet at the top.  That is to say, all the commands appear to be listed as available in the command pallet.  But as mentioned earlier, invoking any MicroPico command from the command pallet appears to do nothing.

dt78q, 2023-11-05:
I had the same issue on LMDE-5 (Linux x64 5.10.0-25-amd64) but 3.4.1 is now working since I upgraded to LMDE-6 (Linux x64 6.1.0-13-amd64)

zopp99, 2023-11-05:
I appreciate the insight. I'm not sure I mentioned prior but I run micropico on Raspberry Pi OS which is kept updated regularly via apt update and apt get. I believe all software is latest versions. Thanks

paulober, 2023-11-05:
@dt78q and @zopp99 Thanks for the details. So it seems like the precompiled python script is causing the issue. 
To prof this assumption @zopp99, can you please try to run:
```
cd ~/.vscode/extensions/paulober.pico-w-go-3.4.1-linux-arm64/dist/scripts/wrapper_linux_arm64/
./wrapper_macOS_arm64.bin --scan-ports
```
(with your Pico connected)

zopp99, 2023-11-05:
@paulober, I could not find 'wrapper_macOS_arm64.bin' so was unable to execute your exact request.  I assumed you may have meant 'wrapper_linux_arm64.bin'.  Following is the command executed and the results of that command with Pico w connected.  I hope this helps.  Thx

pi@raspberrypi:~/.vscode/extensions/paulober.pico-w-go-3.4.1-linux-arm64/dist/scripts/wrapper_linux_arm64 $ ./wrapper_linux_arm64.bin --scan-ports

./wrapper_linux_arm64.bin: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by ./wrapper_linux_arm64.bin)
./wrapper_linux_arm64.bin: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./wrapper_linux_arm64.bin)
./wrapper_linux_arm64.bin: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./wrapper_linux_arm64.bin)
./wrapper_linux_arm64.bin: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./wrapper_linux_arm64.bin)

zopp99, 2023-11-05:
./wrapper_linux_arm64.bin: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by ./wrapper_linux_arm64.bin)
./wrapper_linux_arm64.bin: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./wrapper_linux_arm64.bin)
./wrapper_linux_arm64.bin: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./wrapper_linux_arm64.bin)
./wrapper_linux_arm64.bin: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./wrapper_linux_arm64.bin)

paulober, 2023-11-06:
@zopp99 Yes, that explains the issue. It seems like your installation has an outdated version of `glibc`. That's not a problem for the binaries on your OS or for binaries compiled on your system, but it is for some compiled on a system with a newer version of `glibc`.
```
ldd /bin/ls | grep libc.so
```
There you should find a file called `libc.so.X` where X is the version of your `glibc` => `v2.X`

The only solution I found out, is to update your OS (I guess it was `apt dist-upgrade`).

Though I first will search if I could add more backwards compatibility to the Linux arm64 binary.

zopp99, 2023-11-06:
I am certainly a rookie on unix / debian / Raspberry Pi, but I think I follow you.  Your comment makes sense.  I have fully updated Raspberry PI OS with 'apt full-upgrade'.  'ldd --version' returns glibc 2.31.

I did a bit of internet detective work and THINK(??) I understand the current state of affairs.  There are currently two version of 'Raspberry PI OS' supported.

** The older Raspberry PI OS version is called 'Bullseye' and is based on Debain 11 and GCC 10.2.1.  It has been available and maintained since 12/2021.  It is still supported and widely used.  The 'Bullseye' release appears to upgrade to glibc 2.31.  I understand that glibc under 'Bullseye' can not stably by upgraded beyond 2.31.

** The newer Raspberry Pi OS version is called 'Bookworm' and is based on Debian 12 and GCC 12.2.  It has been available since 10/2023.  I infer that glibc under 'Bookworm' is a more recent version that may be compatible with MicroPico.

It is not possible to simply upgrade Raspberry Pi OS from Bullseye to Bookworm.  Rather a fresh install is required to get to Bookworm - delaying user migration from Bullseye to Bookworm, of course.  

I hope my comments above make sense.  (Please know that this is my best understanding but I could well be way off base so I appreciate your kind excusing me if wrong.)     Options:     1)  Perhaps your last comment about having MicroPico support backward compatibility to glibc 2.31 is practical?   2) I will create a Bookworm bootable drive and run 'ldd --version' and MicroPico to confirm if the very latest Raspberry Pi OS fixes the issue.  (This may take me several days to complete)

I appreciate your feedback and support.  Thanks

paulober, 2023-11-06:
@zopp99 You're right there're two versions of RPi OS right now and the extension currently only
supports glibc from `GLIBC_2.32`. I'll work on a patch ASAP...

Thanks for doing all the research and providing the debug information.

zopp99, 2023-11-06:
So I just whipped up a Bookworm bootable drive.   Bookworm supports glibc 2.36.  I then installed VSCode and Micropico, and Micropico works as expected.  Great!    

Thank you for your outstanding support of my issue.  And best regards to you for an absolutely amazing product in MicroPico.  Ciao.

PR #160 (2023-11-06): Patch v3.4.2
### Changed
- Upgraded to `pyboard-serial-com` `v2.0.4`, Linux `arm64` and `armhf` releases are now backwards compatible with `GLIBC_2.31` (Fixes #159)
- Updated dependencies
- Reload settings before auto-connect to detect settings changes concerning the port
closed 2023-11-06 by paulober