paulober/MicroPico #166: Switch stubs: use `python3 -m pip` instead of `pip`

dpslwk, 2023-11-25:
I don't think stubs.mjs should be using `pip`,
the initial install of the stubs is done in the package.json using `python3 -m pip`
on macOS where python3 is install via brew there is no `pip` the correct exec would be `pip3` but its likely still better to use `python3 -m pip`

paulober, 2023-11-28:
I agree though for example `python3` doesn't exist on windows, there you only have `python(.exe)` depending on your installation. 
Trying both and using the one that exists should fix this.

dpslwk, 2023-11-28:
does that imply that stubs are not even installed correctly on windows as the `postinst` call `python3` will fail?
https://github.com/paulober/MicroPico/blob/0ee67c1909bb62eb0f48a88e43bb4a423be23db2/package.json#L502

paulober, 2023-11-28:
> does that imply that stubs are not even installed correctly on windows as the `postinst` call `python3` will fail?
> 
> https://github.com/paulober/MicroPico/blob/0ee67c1909bb62eb0f48a88e43bb4a423be23db2/package.json#L502

Yes

mvsteen, 2024-04-30:
Could you offer any specific step to get switching stubs (to ESP32) to work then? And am I correct in assuming this would get intellisense working in the on device workspace, not just in the local workspace. Trying to use micropico on a ESP32 "CYD" (cheap yellow display). It's working already, just no autocompletion in the on device workspace.

I am inexperienced in both python and microcontrollers (engineer, lotsa CAD API scripting in VB), so excuse my ignorance.

paulober, 2024-11-18:
Sorry for the delay. No, unfortunately auto-completion not working on device is a Pylance issue and does not seem to be a priority for them. I hope that the stub switching issue has been solved for you with the latest releases?

closed 2024-11-18 by paulober