paulober/MicroPico #167: upload current file in sub-directory

benoitLecuppe, 2023-11-26:
sorry for my bad english i'm an old French ;-)

> uploading a Python file, with is locate in a project subdirectory, upload the file in the root folder on the device not in the subdirectory like upload project do.

## What are the steps to reproduce this issue?
1. have a project with a **subdirectory** and a test.py inside
2. open the file and use commande **microPico : Upload current file to Pico**

## What happens?
the file test.py was uploaded in **root** directory on the device not in **subdirectory**

## Which version of MicroPico are you using?
Published 2022-08-10, 16:43:54
Last released 2023-11-23, 00:07:21
Last updated 2023-11-24, 19:24:12

## Support info
```
Version: 1.84.2 (user setup)
Commit: 1a5daa3a0231a0fbba4f14db7ec463cf99d7768e
Date: 2023-11-09T10:51:52.184Z
Electron: 25.9.2
ElectronBuildId: 24603566
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22000
```

paulober, 2023-12-06:
Good idea though there're some cases that have to be solved first:
Ok, and where would you suggest/like a file stored outside the sync-directory to be uploaded to, root?
And if its within an additional sync root which is child of the main sync-root?

Or would you prefer a short dialog asking if you want to upload it relative to the most direct sync-root the file is in or if you instead want to upload it to the root?

felinh, 2024-01-11:
+1 for this enhancement.
my setting of "Micropico: Sync Folder" is empty, I guess user expects that command "microPico : Upload current file to Pico"   uploads the file onto the device with the similar path as the command "upload project" does.

Thank you for this nice extension, Thonny is really too limited.

benoitLecuppe, 2024-01-12:
Yes, I think in 90% of use cases people want to upload a directory like "upload a project".

My use case:
I have a “/web” folder with tons of static web files and it is taking a long time to upload. So I can't use the "upload project" command.
I also have Python library folders that I upload more often.

Typically, I use the context menu in the left Explorer view.
We have two commands:
- upload the project
- upload a file
Maybe we could have two (or three) commands like "Save" and "Save As...":
the command can apply to a file, a folder, a multiple selection of file and folder.
'**upload**':
1. on file : upload the file respecting the path
2. on folder : uploads the folder and its contents recursively respecting the path
3. on a selection : executes the command on each element of the selection

'**upload files**':
1. on file : same as 'upload'
2. on folder : upload the folder and its files but not its subfolders, respecting the path
3. on a selection : executes the command on each element of the selection.

This command is very useful for copying the root folder without copying subfolders.

'**Upload as...'**: Opens a dialog where we can change the destination and say whether we want the subfolders to be processed recursively or not.

benoitLecuppe, 2024-01-12:
Actualy the drag and drop betwenn project and Remote workspace allows mr to nice get around this problem

sgbaird, 2024-01-26:
> Actualy the drag and drop between project and Remote workspace allows me to nice get around this problem

Just make sure to hold `Ctrl` (Windows) so that it copies rather than moves it.

NJAldwin, 2025-05-21:
This is a serious blocking issue that also affects "run current file"

The "just use Remote workspace" doesn't fix it because many boards don't support that.