paulober/MicroPico #170: confusion over importing modules that python supports but micropython does not.

gilesknap, 2023-12-05:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. put `from typing import Callable1 in a .py file and attempt to run it

## What happens?

There are no static code check errors (so no red squiggles)  but upon execution I see:

ImportError: no module named 'typing'

## What were you expecting to happen?

That I'm able to supply type hints for Callables

## Any other comments?

I'm pretty sure that the typing module is available in micropython but I'm confused by the fact that I see its use in .vscode/Pico-W-Stub/machine.pyi.

This happens with some other modules too. I get no linting errors but when I go to execute the code I then get a runtime error. Perhaps the issue is that I need to uninstall the Pylance Extension?

## Which version of MicroPico are you using?

v3.5.0

## Support info

```
Version: 1.84.2
Commit: 1a5daa3a0231a0fbba4f14db7ec463cf99d7768e
Date: 2023-11-09T10:50:47.800Z
Electron: 25.9.2
ElectronBuildId: 24603566
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux x64 6.2.0-37-generic
```

paulober, 2023-12-05:
@Josverl I think that's something concerning [Josverl/micropython-stubs](https://github.com/Josverl/micropython-stubs)...
_(I'm unable to transfer the issue for some reason)_

Josverl, 2023-12-07:
So technically this is not a bug, but a lack of an assumed feature 🙃

gilesknap, 2023-12-08:
@paulober in general how do I tell if a module that I import really exists in MicroPython I seem to find that linting lets me import many things that don't work at runtime. `import enum` is another example.

paulober, 2023-12-09:
> @paulober in general how do I tell if a module that I import really exists in MicroPython? I seem to find that linting lets me import many things that don't work at runtime. `import enum` is another example. 

Running 'import module' in the vREPL. If it shows you an import error, the module is not available. And generally modules starting with u should be a micropython module.

gilesknap, 2023-12-09:
@paulober OK I guess that works! thanks.

Josverl, 2023-12-13:
>  Does it really enable type checking or just allow you to use code that has type hints?
well - type checing in python is optional - so not real type-checking as in a early-bound language 

also I aim to make the MicroPython type-stubs "not wrong "rather than "always right", and even that is not simple 

I think your  issues `import enum`  is caused by the fact that the stubs also include  a simplified version of the stdlib type-stubs 
and they in turn include , and depend on a the `enum.pyi` type stub.

I maintain an overview of the modules on the different ports and boards :  `enum` is not there 
https://flatgithub.com/Josverl/micropython-stubs/?filename=all_modules.json

if you find other modules that you find misleading , please feel free to raise an issue on https://github.com/Josverl/micropython-stubs/

Josverl, 2023-12-13:
@gilesknap I have published a pre-release of micropython-stdlib-stubs (1.0.1a0)
you can install ( adjust the target location for your setup )
`pip install micropython-stdlib-stubs --pre  --target typings --no-user` 

can you please let me know in the linked issue if this solves your problem , and if you think additional work is needed ?

Josverl, 2023-12-27:
@gilesknap ,  FYI I documented the Why and how :  
https://micropython-stubs.readthedocs.io/en/main/_typing_mpy.html,

gilesknap, 2023-12-27:
@Josverl that looks all good.

paulober, 2024-01-24:
@gilesknap I Fuß the issue can be closed now?

gilesknap, 2024-01-24:
sure, thanks!

closed 2024-01-24 by gilesknap