paulober/MicroPico #171: After creating new project, entries in settings.json appear grayed out

MarByteBeep, 2023-12-05:
I just created a new project as per instructions and when looking at the `settings.json` I noticed a number of settings being grayed out:

[image]

These are the warnings generated by each of the grayed out options:

```
"python.linting.enabled": Unknown Configuration Setting
"python.languageServer": This setting cannot be applied in this workspace. It will be applied when you open the containing workspace folder directly.
"micropico.syncFolder": This setting cannot be applied in this workspace. It will be applied when you open the containing workspace folder directly.
"micropico.openOnStart": This setting can only be applied in user settings in local window or in remote settings in remote window.
```

So I'm guessing these options don't seem to work? I specified 'src' as `syncFolder` but as can be observed in the Pico Remote Workspace, the src folder is synced including the `src` folder itself, which shouldn't be the case.

**Expected**

* No options should be grayed out
* The `"micropico.syncFolder"` setting should sync the contents of said folder so they appear in the root folder on the Pico.

paulober, 2023-12-05:
The settings are greyed out because of the open virtual workspace. As four your other concerns, I'll look into it.

Valitseja, 2024-03-11:
I have similar setup as you @MarByteBeep but the "Upload project to pico" command uploads all files in the project, not just the source. I think if you would add a small file at root it would also be uploaded. 

I would also guess if syncFolder is set to src the content of that folder is uploaded

paulober, 2024-11-18:
Should be fixed with v4.1.0.

closed 2024-11-18 by paulober