paulober/MicroPico #173: Local Python Requirement?

SickProdigy, 2023-12-11:
> I've read in a tutorial elsewhere that local python 3.9 is needed. Is this still the case?

## What are the steps to reproduce this issue?
1. Install vscode
2. Install MicroPico

## What happens?
No extra options under Run

## What were you expecting to happen?
Hoping to see the MicroPico Run functions. 

## Any other comments?
After installing python 3.11 I now see the pico connected at the bottom. I don't see Run > MicroPico though. Ctrl+shift+P then typing pico brought up commands 
https://i.imgur.com/zDaD7uf.png
Then I was able to push to device. It worked. Be nice if the main page said you needed python installed locally. Figured the devices python would be enough. 

## Which version of MicroPico are you using?
v3.5.0

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.85.0 (user setup)
Commit: af28b32d7e553898b2a91af498b1fb666fdebe0c
Date: 2023-12-06T20:48:09.019Z
Electron: 25.9.7
ElectronBuildId: 25551756
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22621
```

paulober, 2023-12-13:
No, Python isn't a requirement anymore as long as you don't want to swap stubs.

paulober, 2023-12-13:
Do you have the value "run" in your `micropico.statusbarButtons` setting (globally or in your local workspace)?

paulober, 2024-01-07:
@SickProdigy I will mark this as resolved since there has been no response. Feel free to reopen it if the problem still exists.

closed 2024-01-07 by paulober