paulober/MicroPico #174: Uploading .bmp files on RPI Pico

NonRidiculousAdjective, 2023-12-15:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.
*Wanted to upload simple bitmap images to my RPI Pico, but the button for uploading them doesn't seem to show up. Upon uploading the whole project, the .bmp file is ignored.*

## Describe the solution you'd like
*Being able to upload .bmp files on the pico.*

## Describe alternatives you've considered
*I know it is technically possible to turn an image into a bytearray and copy/paste that into the code, but that only applies when one has very few of the images, as putting bytearrays of a dozen or more images directly into the program would prove messy. So that's why being able to have the .bmp files on-board would be great.*

paulober, 2023-12-16:
You can add bmp files to the list of filetypes to upload or set `syncAllFileTypes`. Here you can find a short explanation: https://github.com/paulober/MicroPico#extension-settings
In your VS Code settings under Workspace category you can got to MicroPico can there you should find settings for it.

paulober, 2023-12-20:
Did this solve your issue?

paulober, 2024-01-07:
@SickProdigy I will mark this as resolved since there has been no response. Feel free to reopen it if the problem still exists.

closed 2024-01-07 by paulober

NonRidiculousAdjective, 2024-01-12:
> @NonRidiculousAdjective Did this solve your issue?

Yes it did! Thank you very much, apologies for the late response.