paulober/MicroPico #175: Kingsoft virus scanner triggers Malware alert when running project config

tkohlber, 2023-12-23:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. go to virustotal.com
2. upload the binary that's being executed during VS Code project set up, e.g. at /Users/$USER/.vscode/extensions/paulober.pico-w-go-3.5.0-darwin-x64/dist/scripts/wrapper_macOS_amd64/wrapper_macOS_amd64.bin
3. run analysis

## What happens?
All but one virus scanner report "Undetected", except for Kingsoft, which reports "Script.Ks.Malware.6977"

## What were you expecting to happen?
No findings, i.e. all results to be "Undetected"

## Any logs, error output, etc?

https://www.virustotal.com/gui/file/370e7c622615ddd18d05d7aede5d3b5682c6c118b07b4eb9267cdc4f4ef5001c?nocache=1

## Any other comments?

Has anyone else had this issue? 
Is this a false positive?

Do I need to execute the project set up, or can that be done manually, e.g. by manually creating a .vscode/settings.json file?

## Which version of MicroPico are you using?
v3.5.0

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.85.1 (Universal)
Commit: 0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2
Date: 2023-12-13T09:48:06.308Z
Electron: 25.9.7
ElectronBuildId: 25551756
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Darwin x64 23.2.0```

Josverl, 2023-12-23:
That is the signature of a 5 year old WordPress virus. Unlikely that that would be undetected by all other vendors.

I'd still check it, but i think there is a a high chance of a false positive here.

Also Microsoft does scan all extensions and updates. 

https://code.visualstudio.com/docs/editor/extension-marketplace#_can-i-trust-extensions-from-the-marketplace

tkohlber, 2023-12-25:
Thanks for the context. Looks very much like a false positive given that. 

Thanks for looking into it. Wanted to install it on my work machine and be sure.

paulober, 2023-12-26:
THanks @Josverl, that's right it's a false positiv. I had to fight them since compiling the python scripts to remove the Python requirement. I'll get in touch with Kingsoft if possible to clarify this.

paulober, 2023-12-26:
> Also Microsoft does scan all extensions and updates.
> 
> https://code.visualstudio.com/docs/editor/extension-marketplace#_can-i-trust-extensions-from-the-marketplace
> 

Interesting, didn't know that.

paulober, 2023-12-26:
Ok, so it seems to be a bit harder to report this false positive as it's a Chinese company and some people online say: "Kingsoft is no longer tested by AV-Comparatives and by AV-TEST since the mid-2010's. It appears that they no longer exist or [no longer make antivirus/security software](https://malwaretips.com/threads/whatever-happened-to-kingsoft-antivirus-2012-en-version.93737/), at least in English."

paulober, 2023-12-26:
So I don't really want them to have my mail address so I won't report this false positive and I guess your company isn't using Kingsoft, aren't they?

closed 2023-12-26 by paulober

tkohlber, 2023-12-28:
Thanks for the thorough follow-up. Indeed, Kingsoft doesn't seem to provide AV scanning software any longer. Strange that it's still included in the Virustotal suite, which my employer uses (among others). 
I guess I'll have to install your great extension on my personal machine.