paulober/MicroPico #178: How to contribute / Some bugs

webdeb, 2023-12-27:
Hi @paulober,
I've tested the `importOnConnect` feature, and it did't work in my case. (I've just added some prints without anything else)

However, I would like to try find the problem with the extension on my local environment with my own setup. Because I think there is another issue which is related to upload and reset. Could you please provide some details of how to contribute?

The image is related to my issue, the uploading bar is stuck, **however the upload does work fine**! But after a while (10-20sec), the system hangs and there is also an python error. I guess this is related to reset, or terminal reset.

[image]

paulober, 2023-12-28:
Strange.. I did see some people having strange issues like that tough I couldn't fix them yet as I can't test on a machine having this issues. 
Thanks for helping. Here you can find some details on how to contribute: https://github.com/paulober/MicroPico/blob/main/CONTRIBUTING.md
After the setup and login in you local npm with a GitHub access token (with the permission to read packages) you can start debugging my just hitting the Run & Debug button in the vscode debug panel.

paulober, 2023-12-28:
The import on connect only accepts module names though you should be able to directly add a print statement by doing `amodule; print("My stuff")`.

webdeb, 2023-12-28:
Thank you I'll check it out!

webdeb, 2023-12-28:
@paulober ok, setup is ready, I started the extension debugger and opened a project. I can see the newly added micropico icon in the side bar, but all other commands are missing, also short-keys for uploading etc respond with smt like "no command micropico.upload found"

paulober, 2023-12-28:
> @paulober ok, setup is ready, I started the extension debugger and opened a project. I can see the newly added micropico icon in the side bar, but all other commands are missing, also short-keys for uploading etc respond with smt like "no command micropico.upload found" 

Try to run `npm run compile` once before executing the debugger.
And for other errors, you can find them in the output and debug panel in the source vscode instance and the output of the extension host in the debug vscode instance

closed 2023-12-28 by webdeb