paulober/MicroPico #179: Board stuck after upload

webdeb, 2023-12-28:
After some debugging, I found that the `reset.soft.listen` command gets a response from the board and tries to write it into the terminal, which is at least wrong formatted..

After the selected line is executed where data is the text in the hover tooltip the boad is stuck.

[image]

'\r\nb'\\r\\nMPY: soft reboot\\r\\nStarting...\\r\\nInputs created\\r\\nsixmix_program.py\\r\\nAdd custom programs\\r\\n'\n!!PyboardError!!\ncould not enter raw repl\n'

`!!PyboardError!!\ncould not enter raw repl` is an error message from pyboard.py which is defined at several places, I would propose to first add numbers to those lines, so it could be analyzed more precisely.

However, when this line is not executed: did a simple check (not a real fix) that part of the error is gone

```
          if (!data.includes("PyboardError")) {
            terminal?.write(data);
          }
```

There seems to be an uncaught exception in `pyboard-serial` when doing a terminal.write at certain conditions. Cannot investigate it further as the code for it is packaged and minified.

paulober, 2023-12-30:
Thanks for debugging. The source-code for pyboard-serial can be found in https://github.com/paulober/pyboard-serial-com

Do you also have this error with reset.soft.listen with an empty board and a new project where the main python program only includes a print for example?

paulober, 2026-09-14:
This was reported against MicroPico 3.x, which used a connection backend that version 4 replaced completely. If you still see this on the latest version, please open a new issue. Thanks!

closed 2026-09-14 by paulober