paulober/MicroPico #184: Release v3.7.0 never finishes loading in VSCode when extension is activated

Quantum56, 2024-01-18:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Install / update MicroPico to 3.7.0 via the VSCode extension marketplace
2. Open a Micropython project
3. Attempt to connect to a Pico via `> MicroPico: Connect`

## What happens?
The extension never finishes loading

## What were you expecting to happen?

The extension should load (eventually)

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

2024-01-18 18:25:27.331 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onFileSystem:pico'
2024-01-18 18:25:27.958 [error] Error: spawn c:\Users\Notcheroo\.vscode\extensions\paulober.pico-w-go-3.7.0-win32-x64\dist\scripts\wrapper_win32\wrapper_win32_amd64.exe ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:476:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:82:21)

## Any other comments?
N/a

## Which version of MicroPico are you using?
`v3.7.0`

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.85.2 (user setup)
Commit: 8b3775030ed1a69b13e4f4c628c612102e30a681
Date: 2024-01-18T06:40:10.514Z
Electron: 25.9.7
ElectronBuildId: 26354273
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22621
```

shochwald-maptek, 2024-01-19:
+1
Had to downgrade to v3.6 to resume my work.

dvdmllr, 2024-01-19:
+1 
The update left me unable to connect to a Raspberry Pi Pico in VSCode. Downgrading to 3.6.0. fixed the issue for me.

ChrisChoke, 2024-01-19:
on Ubuntu Desktop i dont see se error message above but these

```
2024-01-19 17:09:18.365 [info] [Activator] Connection to wrapper successfully established
2024-01-19 17:09:24.502 [error] [Activator] Pyboard exited with code 1
```

would love to help debugging but dont know what i can do at the moment.

Chris

timopomer, 2024-01-19:
same here, 3.6 still works

ChrisChoke, 2024-01-19:
For me v3.5 works but not above. i am on ubuntu desktop 22.04.3 LTS.
i used original raspberry pico w for testing.

are you all on windows guys?!

MarcoMiano, 2024-01-19:
Windows 11 10.0.22631
vscode 1.85.2

v.3.7 doesn't work even with all others extensions disabled (except the required one)...
it sort of worked one time just configured the project and then nothing... no connect no nothing...

tompaulick, 2024-01-20:
Same problem here - Both on Ubuntu arm64 and Windows 11 x86.  Latest version results in "activating extensions" and then quits.  nothing happens.

shochwald-maptek, 2024-01-20:
I'm on WIndows 11.

paulober, 2024-01-20:
That's because of AV. It should be fixed if you update the virus definitions of Defender. Microsoft already removed the false positive and others follow. Sorry for the inconvinience.

paulober, 2024-01-20:
https://github.com/paulober/MicroPico/releases/tag/v3.7.1

closed 2024-01-20 by paulober

ju-gem, 2024-01-22:
Sorry to rain on your parade, but this is still happening on my machine. :)

paulober, 2024-01-22:
> Sorry to reopen this issue, but this is still happening on my machine.

Np, on which platform are you experiencing this issue?
If it's linux, what does the output of `ldd --version` on your system look like?

shochwald-maptek, 2024-01-22:
I just upgraded to 3.7.3. I see "Activating Extensions..." spin for a while on the status bar, then the extension fails to load.
Looking at Extension Host > Output I see:

`2024-01-23 09:18:02.276 [error] Error: spawn c:\Users\scott\.vscode\extensions\paulober.pico-w-go-3.7.3-win32-x64\dist\scripts\wrapper_win32\wrapper_win32_amd64.exe ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:476:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
`

I guess this is the same issue. Let me know if you want a new issue created.

paulober, 2024-01-23:
@shochwald-maptek It should be the same as https://github.com/paulober/MicroPico/issues/189. Defender deleted the file cause of outdated virus definitions. It should Microsoft maybe about 12h to fix this issue. (if it's the same issue as in the linked one) But just to know,  what output do you get for `ls c:\Users\scott\.vscode\extensions\paulober.pico-w-go-3.7.3-win32-x64\dist\scripts\wrapper_win32` in powershell?

shochwald-maptek, 2024-01-23:
```
    Directory: C:\Users\scott\.vscode\extensions\paulober.pico-w-go-3.7.3-win32-x64\dist\scripts\wrapper_win32

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a----        23/01/2024   9:16 AM        5162776 libcrypto-3.dll
-a----        23/01/2024   9:16 AM          39696 libffi-8.dll
-a----        23/01/2024   9:16 AM        5785880 python311.dll
-a----        23/01/2024   9:16 AM          30488 select.pyd
-a----        23/01/2024   9:16 AM        1141016 unicodedata.pyd
-a----        23/01/2024   9:16 AM         119192 vcruntime140.dll
-a----        23/01/2024   9:16 AM          84760 _bz2.pyd
-a----        23/01/2024   9:16 AM         124696 _ctypes.pyd
-a----        23/01/2024   9:16 AM         253720 _decimal.pyd
-a----        23/01/2024   9:16 AM          65304 _hashlib.pyd
-a----        23/01/2024   9:16 AM         159000 _lzma.pyd
-a----        23/01/2024   9:16 AM          79640 _socket.pyd

```

shochwald-maptek, 2024-01-23:
Ah so it is! I will try it again tomorrow.
Thanks

[image]

paulober, 2024-01-23:
@shochwald-maptek Microsoft just released new definitions for defender that fix this issue. Though you seem to require a reinstall of the extension as defender removed the file on you system.