paulober/MicroPico #188: Micropico stopped working after version 3.6.0  on Raspberry Pi 4 running Bookworm OS   (Similar to issue #159??)

zopp99, 2024-01-21:
> Please provide a clear and concise description of the bug along with logs
MicroPico versions 3.7.0 and later do not appear to do anything when MicroPico commands are executed. (Downgrading to MicroPico version 3.6.0 works fine).

Beginning at version 3.7.0, I no longer see Micropico 'buttons' on the bottom status bar. Executing commands directly via the command pallet briefly shows a 'activating extensions' message but nothing happens. There are no messages in the MircroPico log except for one initial line showing 'Installed stubs are already up to date!'.

I am running Raspberry Pi Bookworm on a Pi4 SBC.  All update from Raspberry Pi are upplied / i.e., system is up to date.

## What are the steps to reproduce this issue?
1. Install MicroPico 3.7.0 through 3.7.3 (or latest rlse)
 2. Navigate to any prior valid micropico folder / upython program
 3. Execute any micropico command. -> nothing appears to happen

## What happens?
No MicroPico 'buttons' on the bottom vscode status bar. When executing commands directly via command pallet, nothing appears to happen. No log messages are generated.

## What were you expecting to happen?
MicroPico buttons should be displayed at bottom of vscode status bar. MicroPico commands should be executed.

## Any logs, error output, etc?
2024-01-21 17:28:05.106 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2024-01-21 17:28:05.977 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2024-01-21 17:28:06.013 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.micropico'
2024-01-21 17:28:06.608 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2024-01-21 17:28:06.814 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2024-01-21 17:28:07.998 [error] Error: spawn /home/williamziemniak/.vscode/extensions/paulober.pico-w-go-3.7.3-linux-arm64/dist/scripts/wrapper_linux_arm64/wrapper_linux_arm64.bin ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:476:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
2024-01-21 17:28:10.823 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2024-01-21 17:28:15.845 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2024-01-21 17:28:15.869 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2024-01-21 17:28:41.877 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python'
2024-01-21 17:28:43.186 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2024-01-21 17:28:43.817 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2024-01-21 17:28:47.061 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'

## Any other comments?
The symptoms are very similar to my prior bug issue #159.  Perhaps the resolution can be found with issue #159.  Thank you

## Which version of MicroPico are you using?
version 3.7.3

## Support info
```
Version: 1.83.0
Commit: e7e037083ff4455cf320e344325dacb480062c3c
Date: 2023-10-03T16:12:06.378Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux arm64 6.1.0-rpi7-rpi-v8
```

paulober, 2024-01-22:
@zopp99 Thanks for reporting.
What output do you get for following commands?
cd ~/.vscode/extensions/paulober.pico-w-go-3.7.3-linux-arm64/dist/scripts/wrapper_linux_arm64/
./wrapper_linux_arm64.bin --scan-ports

zopp99, 2024-01-22:
the CD works fine.  And in that directory there is a .bin for wrapper_linux_arm64.    However, when I execute your second command I receive the following "bash: ./wrapper_linux_arm64.bin: cannot execute: required file not found".

I hope this might help.  Thanks

paulober, 2024-01-22:
> the CD works fine. And in that directory there is a .bin for wrapper_linux_arm64. However, when I execute your second command I receive the following "bash: ./wrapper_linux_arm64.bin: cannot execute: required file not found".
> 
> I hope this might help. Thanks

It does. Seems like my linux arm64 messed up the build process of the binary. The issue is fixed with hot fix v3.7.4.
Sorry for the inconvenience.

closed 2024-01-22 by paulober

zopp99, 2024-01-22:
You're the best.  The tool is amazingly helpful, and support is top shelf.  Best Regards.

commit 58ee1ad3 (2024-01-22): Fix #188, Wrapper not running on linux arm64
PR #190 (2024-01-22): Hotfix v3.7.4
### Changed
- Upgraded to `pyboard-serial-com` `v3.0.6` (Fixes #188)