paulober/MicroPico #189: Today wrapper_win32_amd64.exe reported as malware by Microsoft Defender

g3zod, 2024-01-22:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1.Windows 11 reported this on its own.

## What happens?
Windows Defender quarantines wrapper_win32_amd64.exe and reports:
Detected: Trojan:Win32/Wacatac.B!m
Status: Quarantined
Date: 22/01/2024 18:09
Details: This program is dangerous and executes commands from an attacker.
Affected items:
file: C:\Users\g3zod\.vscode\extensions\paulober.pico-w-go-3.7.3-win32-x64\dist\scripts\wrapper_win32\wrapper_win32_amd64.exe

## What were you expecting to happen?
Not to be detected as malware.

## Any logs, error output, etc?
See "What happens?" above.

## Any other comments?
Had no problems until today.

## Which version of MicroPico are you using?
3.7.3

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.85.2 (user setup)
Commit: 8b3775030ed1a69b13e4f4c628c612102e30a681
Date: 2024-01-18T06:40:10.514Z
Electron: 25.9.7
ElectronBuildId: 26354273
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22621
```

paulober, 2024-01-22:
@g3zod Thanks for reporting. Microsoft will release an update to the Defender AV definitions within the next 12h that should fix the issue.

g3zod, 2024-01-23:
Great news - many thanks.

paulober, 2024-01-23:
Microsoft just released new definitions for defender that fix this issue

closed 2024-01-23 by paulober

g3zod, 2024-01-26:
All good after regular quick scans and a full one - many thanks again.