paulober/MicroPico #191: Clicking on "Toggle Pico-W-FS" on status bar removes all MicroPico buttons from status bar - and cannot be turned back on again

wright-david, 2024-01-25:
## What are the steps to reproduce this issue?
1.  Clicking on "Toggle Pico-W-FS" on status bar remove MicroPico buttons
2.  via Command Palette try to re-enable this by running "MicroPico: Toggle Virtual File System (reloads UI and closes existing vREPLs)" 

## What happens?
original buttons on status bar do not reappear 

## What were you expecting to happen?
 original buttons on status bar should reappear 

## Any other comments?
Required uninstall/reinstall of extension to restore status bar (disable/enable made no difference)

## Which version of MicroPico are you using?
v3.7.4

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.85.2 (user setup)
Commit: 8b3775030ed1a69b13e4f4c628c612102e30a681
Date: 2024-01-18T06:40:10.514Z
Electron: 25.9.7
ElectronBuildId: 26354273
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22631
```

paulober, 2024-01-26:
Thanks for reporting this issue. Do you get any output in 'Output> Extension Host` when this issue happens?

wright-david, 2024-01-27:
Thanks also for your prompt response.

Although the other day I couldn't seem to stop this from happening, unfortunately today I can't reproduce it.

paulober, 2024-01-27:
Please reach back to this issue if you experience this issue again.

wright-david, 2024-02-14:
Update 14-Feb-2024:   have not experienced this since initial report (and have probably finished playing with Pico for time being).

paulober, 2024-02-15:
Ok, feel free to reopen if you should encounter this issue again.

closed 2024-02-15 by paulober

breezin320, 2024-03-24:
I have this issue.  I'm using MicroPico v3.7.5.  On Windows 11, VSCODE: Version: 1.87.2 (user setup)
If I toggle Pico-W-FS, the bottom line buttons (Run, Reset, All commands, etc) disappear.  There is continuous cycling of Pico W Remote Workspace, with no other output, in the Workspace section on the left.  The only way to clear is as reported, uninstall/reinstall extension a few times.   It might be there are no files on the Pico W (is that possible?).

More than happy to try any suggestions to troubleshoot.....

thanks for your help!
dave

breezin320, 2024-03-24:
Apologies for the second post.  To add to my previous comment, here's a screen grab of what I see:

[image]

breezin320, 2024-03-24:
One more data point....If I right click on the Pico(W)Remote Workspace, and select Remove Folder from Workspace, it stops the cycling and the prompts reappear at the bottom again.

thanks!
dave