paulober/MicroPico #193: Ability to disable repeated notifications of "Connection to Pico lost"

sgbaird, 2024-01-29:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.

The following message (bottom-right corner) pops up over and over again when I don't have a Pico W connected. The message is pretty distracting, especially since it "pops up" again, instead of just staying there. If I click "X" on a notification, it just comes back again.

[image]

Even when I click "Turn notifications off" it still pops up with the message:

[image]

## Describe the solution you'd like

Create an option to disable these notifications/popups. In my case, it's pretty clear whether the Pico W is connected vs. disconnected based on the status bar icon, though I see how it would be generally useful to know if it has disconnected when that isn't the user's intention.

## Describe alternatives you've considered

I've ended up disabling and reenabling the entire extension on a per-workspace basis.

vbakke, 2024-02-13:
I agree. This is a really annoying repetitive warning. One warning when I disconnect the cable is fine. But it reappears every 7.5 seconds.

I use MicroPico v3.7.4 on a Windows VS Code 1.86.1

sgbaird, 2024-02-13:
@vbakke Since I'm still doing a fair bit of development off-and-on with the Pico W, I've decided to just connect one to make the warning go away.. 🫠

vbakke, 2024-02-14:
One way of solving it, I guess. 

Hard for me to know if the problem (or the solution) lies in [MicroPython (blame code)](https://github.com/paulober/MicroPico/blame/887d075ac4847db118be07327d53c7c3c31d5c0c/src/activator.mts#L1283), or if it is the PyBoard's [Pyboardrunner firing onExit](https://github.com/paulober/pyboard-serial-com/blame/2a9a07c098de624a015ca1dc9e877dffe4c68ebc/src/pyboardRunner.ts#L259) non-stop.

Hopefulle someone else managed to find the cause. 🙏

migs35323, 2024-02-20:
Just installed this extension, and come to look for this, when i open code it shows multiple warnings of the connection lost, i believe there should be improvements on this, it makes vscode a bit slower and awkard in general when trying to use other unrelated projects

paulober, 2024-02-21:
These unnecessary notifications should be fixed with the next patch. Thanks for the feedback.

closed 2024-02-22 by paulober

vbakke, 2024-02-26:
Looks good when I tested just now. :) 👍

commit 63e32a16 (2024-02-21): Fix #193, Notifications cleanup
PR #205 (2024-02-22): Patch v3.7.5
### Changed
- Upgraded to `pyboard-serial-com` `v3.0.7`,
- Updated dependencies
- Fixed #202, Switch stubs Windows shell error
- Fixed #193, Notifications cleanup
- Min VS Code version is now `v1.86.2`
- Updated stubs to MicroPython v1.22.1 stable (Thanks to @Josverl [`micropython-rp2-rpi_pico_w-stubs`](https://github.com/Josverl/micropython-stubs/releases/tag/v1.22.0))