paulober/MicroPico #194: Deprecation Warning: Buffer() is deprecated due to security and usability issues.

Redsandro, 2024-01-30:
### Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.

## What are the steps to reproduce this issue?

```bash
$ codium --install-extension paulober.pico-w-go
Installing extensions...
Installing extension 'paulober.pico-w-go'...
(node:53057) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
Extension 'paulober.pico-w-go' v3.7.4 was successfully installed.
```

## What happens?

> DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.

## What were you expecting to happen?

No warnings

## Any other comments?

I am new, first time user, literally just installed MicroPico, so I am not sure if this deprecation is from MicroPico or a dependency of MicroPico.

## Which version of MicroPico are you using?

3.7.4

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.85.2
Release: 24019
Commit: c8a36a69377856369f139f39ee56dd3838130fbf
Date: 2024-01-19T18:30:44.948Z
Electron: 25.9.7
ElectronBuildId: undefined
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Linux x64 5.15.0-91-generic
```

paulober, 2024-01-30:
Hi, I do get this warning also when working with other extensions and it seems like it's a vscode issue or and issue in one of the essential libraries. I may look into it if I have time. But for now it's normal and shouldn't affect usability.

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around its dependencies and build. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.

closed 2026-10-22 by github-actions[bot]