paulober/MicroPico #195: The ignore folder for upload doesn't appear to work.

EdwardHanna, 2024-01-31:
I have a folder called unused that I don't want uploaded, but I've tried adding /unused, \unused, **/unused and other combinations to the ignore list and it always uploads the folder anyway.

Is this a bug or am I doing it wrongly?

Ver 3.7.4

ChrisChoke, 2024-04-08:
Hey,

maybe its the same as #214.
did you toggled the "pico-w-fs" button for getting the filesystem as folder in your workspace?! then vscode changed your workspace to multi-folder-workspace and the settings.json in .vscode folder will not respected.

Chris

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around uploading and syncing files. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.

closed 2026-10-22 by github-actions[bot]