paulober/MicroPico #197: Problems after re-connecting usb

jkozniewski, 2024-02-02:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Run the code via "run" command after successfully connected to pico 
2. Disconnect usb, and connect again. wait for automatic re-connect (sometimes there is a sequence "connected" / "disconnected" / "connected" )
3. Try to run program again

## What happens?
Despite the "Pico Connected" info in status bar trying to run any command does nothing, no error, just nothing happens 

## What were you expecting to happen?
All commands should work as expected after re-connecting usb and while "Pico Connected" status is displayed.

## Any logs, error output, etc?
No explicit errors shown in OUTPUT, there is following info generated after opening VS Code, successfully connecting to pico, running code, then disconnecting usb and re-connecting:
```
2024-02-02 16:08:23.442 [info] [Stubs] Installed stubs are already up to date!
2024-02-02 16:08:23.466 [info] [Activator] Connection to wrapper successfully established
2024-02-02 16:12:41.322 [error] [Activator] Pyboard exited with code 3
2024-02-02 16:12:41.340 [error] [Activator] Pyboard exited with code 1
2024-02-02 16:12:47.434 [error] [Activator] Pyboard exited with code 1
2024-02-02 16:12:48.875 [info] [Activator] Connection to wrapper successfully established
2024-02-02 16:13:00.890 [error] [Activator] Pyboard exited with code 1
2024-02-02 16:13:01.391 [info] [Activator] Connection to wrapper successfully established
```

## Any other comments?
I'm on Mac Book Pro M3 Max; Sonoma 14.1.1

It seems there is some issue with ports being locket / not properly released since after doing:
`ps -ef | grep /dev/cu.usbmodem21201`
I get (even after closing VS Code):
```
501 82229 1   0  4:13PM ??         0:00.03 /Users/jkozniewski/.vscode/extensions/paulober.pico-w-go-3.7.4-darwin-arm64/dist/scripts/wrapper_macOS_arm64/wrapper_macOS_arm64.bin -d /dev/cu.usbmodem21201 -b 115200
```
Then after killing the process via `kill` command pointing to PID of the process above I'm usually able to run / upload / etc again.

Aother cue to sometihng wrong with ports being blocked is that when I launch Thonny after MicroPico stops working I get the message there such as:
```
Unable to connect to /dev/cu.usbmodem21201: [Errno 35] Could not exclusively lock port /dev/cu.usbmodem21201: [Errno 35] Resource temporarily unavailable

If you have serial connection to the device from another program, then disconnect it there first.

Process ended with exit code 1.
```

Even if VS Code is closed before launching Thonny 

## Which version of MicroPico are you using?
MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico W with RP2040

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.86.0 (Universal)
Commit: 05047486b6df5eb8d44b2ecd70ea3bdf775fd937
Date: 2024-01-31T10:29:15.765Z
Electron: 27.2.3
ElectronBuildId: 26495564
Chromium: 118.0.5993.159
Node.js: 18.17.1
V8: 11.8.172.18-electron.0
OS: Darwin arm64 23.1.0
```

jkozniewski, 2024-03-03:
any updates on this matter - seems to be possibly related to https://github.com/paulober/MicroPico/issues/199 ?

jkozniewski, 2024-03-29:
Hi, @paulober sorry for the delayed reply but I had to postpone working on the project a bit, and now getting back...
I've replaced the folder but now Mac Os whines about the code not being signed:`“wrapper_macOS_arm64.bin” cannot be opened because the developer cannot be verified.` so not sure how can I work around that.. also if that potentially fixes the problem, can you just deploy it in new version of the extension ?

Also - now (I mean before I've attempted replacing the folder) occasionally it stuck during project upload, once it even bricked the Pico and I had to put the fresh uf2 again...

jkozniewski, 2024-05-25:
I have replaced the contents of a wrapper with the files you have provided some time ago and it seemed to help for a while - though it still wasn't perfect in terms of connection/disconnecting bug. But after getting back to the project I have to report that the problem persists 
- not sure if that's because an automatic update of a extension (v3.7.6) overwrote the wrapper lib to a worse version again ?
 
Anyway the connection to the board is super finicky - a rather random sequence of disconnecting the usb, relaunching VSCode, connection again etc. end even sometimes just launching Thonny in between gets problem sorted for a while.

Let me know how can I help to investigate the issue further to get thoing working in a stable manner.
Thanks !

jkozniewski, 2024-05-25:
oh and btw - I ran the command you have mentioned before (`lsof /dev/cu.<insert the port name of your board here>`) while experiencing the issues and got following output:
```
COMMAND     PID        USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
UltiMaker 29003 jkozniewski   67u   CHR   9,19    0t182 4945 /dev/cu.usbmodem21201
wrapper_m 70385 jkozniewski    3u   CHR   9,19    0t107 4945 /dev/cu.usbmodem21201
```
Curious thet there is this Ultimaker thing - that's probably because I had Cura launched, nevertheless quitting it didn't really help and got just "wrapper" entry when I've re-ran the cmd.

jkozniewski, 2024-05-26:
Ok and now after problems with connecting to Pico and quitting VSCode the `lsof /dev/cu.` gives:

```
COMMAND    PID        USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
wrapper_m 3050 jkozniewski    3u   CHR    9,1    0t486 5063 /dev/cu.usbmodem21201
```

So it seems the process is not properly killed / not releasing usb port properly ?

jkozniewski, 2024-06-15:
Hello - any chance for making a serial connection with pico more stable ?
I still have pretty bad experience and it kind of semi randomly works better or worse - though Thonny seems much more reliable so I really doubt it's anything hardware related. If I can help any further with debbuging just let me know !

paulober, 2024-09-12:
> Hello - any chance for making a serial connection with pico more stable ? 

Yes, with v4. It's finally out and has many improvements concerning stability. Does v4 finally solve your issues?

jkozniewski, 2024-09-16:
Hi - Thanks Paul - indeed I got back working with Pico Pi recently and it seems to be much more stable than before.
Still not perfect - it can get occasionally stuck (and then re-connecting in Thonny and re-raluching VS Code is proven method to get back on track) but I can feel there was an improvement ! Much appreciated. I'll keep you posred if I'll find any reproducible errors.

paulober, 2026-09-14:
This was reported against MicroPico 3.x, which used a connection backend that version 4 replaced completely. If you still see this on the latest version, please open a new issue. Thanks!

closed 2026-09-14 by paulober