paulober/MicroPico #199: Issues connecting on a MAC toggling from Pico Connect to Pico Disconnected

markt22, 2024-02-06:
> Status bar toggles between Pico Connected and Pico Disconnected 

**Output logs:**
2024-02-06 12:54:04.792 [info] [Stubs] Installed stubs are already up to date!
2024-02-06 12:54:04.794 [info] [Activator] Connection to wrapper successfully established
2024-02-06 12:54:10.857 [error] [Activator] Pyboard exited with code 1
2024-02-06 12:55:48.722 [info] [Stubs] Installed stubs are already up to date!
2024-02-06 12:55:48.724 [info] [Activator] Connection to wrapper successfully established
2024-02-06 12:56:01.109 [error] [Activator] Pyboard exited with code 1
2024-02-06 13:13:48.287 [info] [Activator] Connection to wrapper successfully established
2024-02-06 13:14:00.665 [error] [Activator] Pyboard exited with code 1

**Following earlier issues debugging**
$ ./wrapper_macOS_arm64.bin -d /dev/tty.usbmodem112301
b'connected\r\n'
!!PyboardError!!
could not enter raw repl

**Tried listen2pico.py, seems to be working**
python3 listen2pico.py /dev/tty.usbmodem112301
Connecting to pyboard; press Ctrl-C to exit...

^C
Exiting, joining threads...
Done.

## Which version of MicroPico are you using?
I've tried multiple versions latest is:
MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>>

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.86.0 (Universal)
Commit: 05047486b6df5eb8d44b2ecd70ea3bdf775fd937
Date: 2024-01-31T10:29:15.765Z
Electron: 27.2.3
ElectronBuildId: 26495564
Chromium: 118.0.5993.159
Node.js: 18.17.1
V8: 11.8.172.18-electron.0
OS: Darwin arm64 22.3.0
```

paulober, 2024-02-07:
Please try with cu. instead of tty.

markt22, 2024-02-07:
same:
./wrapper_macOS_arm64.bin -d /dev/cu.usbmodem12301
b'connected\r\n'
!!PyboardError!!
could not enter raw repl

markt22, 2024-02-07:
Wrote a simple program that prints "sleeping 1" every second. Used listen2pico.py to test 

python3 listen2pico.py /dev/cu.usbmodem12301
Connecting to pyboard; press Ctrl-C to exit...
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping ^C
Exiting, joining threads...
Done.

paulober, 2024-02-07:
@markt22 Thanks for testing. So it seems like that for some reason on your system the micropython pyboard tool is unable to enter raw repl mode. It could help if you have a serial console program like minicom and connect to your board and press Ctrl+A and send me the output.

markt22, 2024-02-07:
MPY: soft reboot
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
Sleeping 1
**************** ctrl+a Entered here nothing *****************
Sleeping 1
**************** ctrl+c Entered here, enters raw REPL ******
Traceback (most recent call last):
  File "boot.py", line 12, in <module>
KeyboardInterrupt:
MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>>
raw REPL; CTRL-B to exit
>

paulober, 2024-02-07:
if you first press ctrl-c (maybe twice) and then ctrl-a?

markt22, 2024-02-07:
ctrl-c ctrl-a, gets the 'raw REPL' response
ctrl-a ctrl-c, gets the 'raw REPL' response
ctril-c ctrl-c ctrl-a gets the 'raw REPL' response
But ctrl-a only, the program keeps running.

I downloaded pyboard from micropython tools, it only sends ctrl-a (assuming --no-soft-reset) which doesn't interrupt the running program, also because my program is printing out the serial port read_until never times out.

paulober, 2024-02-07:
ok, thanks for doing all of the testing. So maybe the lib doesn't stop the executing program correctly.

markt22, 2024-02-07:
is your wrapper using pyboard to provide the commincations layer?

paulober, 2024-02-07:
right

markt22, 2024-02-07:
### Now Added a virtual serial tap between the micropython board and wrapper code

**This chunk of data is out of the buffer it is just multiple instances of 'sleeping 1'**  
> 2024/02/07 17:29:33.000710832  length=276 from=1920 to=2195
 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a 53 6c 65 65 70 69 6e 67 20 31 0d 0a

**This is the enter raw sequence from wrapper program** 
< 2024/02/07 17:29:33.000711038  length=5 from=0 to=4
 0d 03 03 0d 01

** Response from micropython board **
> 2024/02/07 17:29:33.000711473  length=34 from=2196 to=2229
 54 72 61 63 65 62 61 63 6b 20 28 6d 6f 73 74 20 72 65 63 65 6e 74 20 63 61 6c 6c 20 6c 61 73 74 29 3a
> 2024/02/07 17:29:33.000711788  length=64 from=2230 to=2293
 0d 0a 20 20 46 69 6c 65 20 22 62 6f 6f 74 2e 70 79 22 2c 20 6c 69 6e 65 20 31 32 2c 20 69 6e 20 3c 6d 6f 64 75 6c 65 3e 0d 0a 4b 65 79 62 6f 61 72 64 49 6e 74 65 72 72 75 70 74 3a 20 0d 0a 72
> 2024/02/07 17:29:33.000711865  length=53 from=2294 to=2346
 61 77 20 52 45 50 4c 3b 20 43 54 52 4c 2d 42 20 74 6f 20 65 78 69 74 0d 0a 3e 72 61 77 20 52 45 50 4c 3b 20 43 54 52 4c 2d 42 20 74 6f 20 65 78 69 74 0d 0a 3e

** Which Translates to** 
> Traceback (most recent call last):
>   File "boot.py", line 12, in <module>
> KeyboardInterrupt:
> raw REPL; CTRL-B to exit
>  >

** From wrapper ctrl-d
< 2024/02/07 17:29:33.000712450  length=1 from=5 to=5
 04

** Response
> 2024/02/07 17:29:33.000712640  length=2 from=2347 to=2348
 4f 4b (OK)

> 2024/02/07 17:29:33.000717211  length=20 from=2349 to=2368
 0d 0a 4d 50 59 3a 20 73 6f 66 74 20 72 65 62 6f 6f 74 0d 0a
(MPY: soft reboot)

**Back to running boot.py**
> 2024/02/07 17:29:33.000812100  length=10 from=2369 to=2378
 53 6c 65 65 70 69 6e 67 20 31
> 2024/02/07 17:29:33.000812170  length=2 from=2379 to=2380
 0d 0a
> 2024/02/07 17:29:34.000812401  length=12 from=2381 to=2392
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:35.000812457  length=12 from=2393 to=2404
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:36.000812708  length=12 from=2405 to=2416
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:37.000812887  length=12 from=2417 to=2428
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:38.000812975  length=12 from=2429 to=2440
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:39.000813093  length=12 from=2441 to=2452
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:40.000813258  length=12 from=2453 to=2464
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:41.000813401  length=12 from=2465 to=2476
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:42.000813521  length=12 from=2477 to=2488
 53 6c 65 65 70 69 6e 67 20 31 0d 0a
> 2024/02/07 17:29:43.000813601  length=12 from=2489 to=2500

###Looking at pyboard.py code
The 04 reset command is sent in (which is default behavior). There is a --no-soft-reset option in the pyboard.py code, which bypasses this code:

    if soft_reset:
            data = self.read_until(1, b"raw REPL; CTRL-B to exit\r\n>")
            if not data.endswith(b"raw REPL; CTRL-B to exit\r\n>"):
                print(f"DEBUG 1 {data}")
                raise PyboardError("could not enter raw repl")

            self.serial.write(b"\x04")  # ctrl-D: soft reset

jkozniewski, 2024-03-03:
maybe its's somehow connected to this issue - https://github.com/paulober/MicroPico/issues/197 ?

paulober, 2024-11-18:
@markt22 Has this problem been solved for you after version 4.0?

paulober, 2026-09-14:
I'm closing this as I couldn't get the details needed to look into it, and a lot has changed since. If it still happens with the latest version, please open a new issue using the bug form. Thanks!

closed 2026-09-14 by paulober