paulober/MicroPico #204: [error] Invalid version for /usr/bin/python

migs35323, 2024-02-20:
Not sure if this issue is related to this extension or it's the python extension itself or its something on my setup

i do have python 3.10

when i open the pico filesystem it detects python 2.7 and shows a bunch of errors:

```log
2024-02-20 21:29:49.420 [error] Invalid version for /usr/bin/python: {"name":"","location":"","kind":"global-other","executable":{"filename":"/usr/bin/python","sysPrefix":"","ctime":1658686354352,"mtime":1633955694000},"searchLocation":{"$mid":1,"fsPath":"/","path":"/","scheme":"file"},"display":"Python 2.7.x","version":{"major":2,"minor":7,"micro":-1},"arch":1,"distro":{"org":""},"source":["path env var"],"id":"/usr/bin/python","detailedDisplayName":"Python 2.7.x"}
2024-02-20 21:29:49.420 [error] Invalid version for /usr/bin/python: {"name":"","location":"","kind":"global-other","executable":{"filename":"/usr/bin/python","sysPrefix":"","ctime":1658686354352,"mtime":1633955694000},"searchLocation":{"$mid":1,"fsPath":"/","path":"/","scheme":"file"},"display":"Python 2.7.x","version":{"major":2,"minor":7,"micro":-1},"arch":1,"distro":{"org":""},"source":["path env var"],"id":"/usr/bin/python","detailedDisplayName":"Python 2.7.x"}
2024-02-20 21:29:53.968 [info] Selected workspace /home/ubuntu/Desktop/pico for creating virtual environment.
2024-02-20 21:29:56.615 [info] Selected workspace / for creating virtual environment.
2024-02-20 21:30:01.916 [info] Selected interpreter /usr/bin/python for creating virtual environment.
2024-02-20 21:36:06.156 [info] Discover tests for workspace name: pico - uri: /home/ubuntu/Desktop/pico
2024-02-20 21:36:06.156 [info] Discover tests for workspace name: Pico (W) Remote Workspace - uri: /
2024-02-20 21:36:06.162 [error] Invalid version for /usr/bin/python: {"name":"","location":"","kind":"global-other","executable":{"filename":"/usr/bin/python","sysPrefix":"","ctime":1658686354352,"mtime":1633955694000},"searchLocation":{"$mid":1,"fsPath":"/","path":"/","scheme":"file"},"display":"Python 2.7.x","version":{"major":2,"minor":7,"micro":-1},"arch":1,"distro":{"org":""},"source":["path env var"],"id":"/usr/bin/python","detailedDisplayName":"Python 2.7.x"}
2024-02-20 21:36:06.162 [error] Invalid version for /usr/bin/python: {"name":"","location":"","kind":"global-other","executable":{"filename":"/usr/bin/python","sysPrefix":"","ctime":1658686354352,"mtime":1633955694000},"searchLocation":{"$mid":1,"fsPath":"/","path":"/","scheme":"file"},"display":"Python 2.7.x","version":{"major":2,"minor":7,"micro":-1},"arch":1,"distro":{"org":""},"source":["path env var"],"id":"/usr/bin/python","detailedDisplayName":"Python 2.7.x"}
2024-02-20 21:38:47.612 [info] Python interpreter path: ./usr/bin/python
2024-02-20 21:38:51.846 [info] Python interpreter path: /bin/python
2024-02-20 21:38:54.263 [info] Python interpreter path: ./usr/bin/python
2024-02-20 21:39:00.876 [info] Python interpreter path: /bin/python
2024-02-20 21:39:09.385 [info] Python interpreter path: ./usr/bin/python
2024-02-20 21:39:16.259 [info] Python interpreter path: /bin/python
2024-02-20 21:39:43.425 [info] Python interpreter path: ./usr/bin/python
2024-02-20 21:39:53.106 [info] Python interpreter path: /bin/python
```

paulober, 2024-02-22:
Where did you find these messages in VS Code?

paulober, 2024-02-22:
I guess either `/usr/bin/python -V` or `/bin/python -V` returns a 2.x version.

paulober, 2026-09-14:
This log comes from the Python extension picking up an old system Python, not from MicroPico, and a lot has changed since. If you still run into problems with the latest version, please open a new issue using the bug form. Thanks!

closed 2026-09-14 by paulober