paulober/MicroPico #206: How can I get a listing of the files already on the Pico?

frohro, 2024-02-29:
#  In Thonny I can see  a list ofo the files on the pico in addition to the files on my laptop.  Can we have this feature too?

## Is your feature request related to a problem? Please describe.
* Maybe an lspico command or something....

## Describe the solution you'd like
*A clear and concise description of what you want to happen. Add any considered drawbacks.*

## Describe alternatives you've considered
*A clear and concise description of any alternative solutions or features you've considered.*

## Teachability, Documentation, Adoption, Migration Strategy
*If you can, explain how users will be able to use this and possibly write out a version the docs. Maybe a screenshot or design?*

frohro, 2024-02-29:
Well, one way that works is to make a file with this inside it and run it.
import os
print(os.listdir())

Is there anything better?

paulober, 2024-03-01:
Opening the MicroPico vREPL (a custom terminal profile) and type your code or just `.ls` and press enter.
Or opening the virtual filesystem/workspace.

paulober, 2026-09-14:
Fixed in [MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0): `.ls` lists the files on the board again (see #315). Thanks for reporting!

closed 2026-09-14 by paulober