paulober/MicroPico #207: MicroPico disconnect won't free serial port

Prototype-Laboratory, 2024-03-06:
## What are the steps to reproduce this issue?
1. open VSCode and connect with the MicroPico extension to the Raspberry Pi Pico
2. disconnect by pressing the "toggle board connection" button or by using the MicroPico Command "Disconnect" 
3. open another serial monitor and try to connect to the Raspberry Pi Pico (won't work) 

## What happens?
Even if I disconnect the Raspberry Pi Pico, the tty port is still occupied by the extension, so using another serial monitor to connect to the Pico won't work

## What were you expecting to happen?
Release the serial port after pressing the "toggle board connection" button to disconnect the Pico.

## Any other comments?
- I am using Micro Python firmware v1.22.1
- Even if I close VSCode, I am not able to connect to the Pico with another serial Monitor. I have to reconnect the USB-Connection.
- Basically I want to write a python program, which is connecting via pyserial to the Raspberry Pi Pico, which won't work, because the Port is still opened.

## Which version of MicroPico are you using?
3.7.5

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.87.0
Commit: 019f4d1419fbc8219a181fab7892ebccf7ee29a2
Datum: 2024-02-27T23:42:16.599Z
Electron: 27.3.2
ElectronBuildId: 26836302
Chromium: 118.0.5993.159
Node.js: 18.17.1
V8: 11.8.172.18-electron.0
Betriebssystem: Linux x64 6.6.16-2-MANJARO
```

jkozniewski, 2024-03-07:
hi - I guess it's related to https://github.com/paulober/MicroPico/issues/197

@Prototype-Laboratory are you on Mac platform (intel / silicon) ? 

@paulober - any chance to look into this ? seems it's pretty common issue already raised in three related issues and it makes working with pico almost impossible, let me know is you need any additional debug data apart from what I've provided in  https://github.com/paulober/MicroPico/issues/197

Prototype-Laboratory, 2024-03-07:
No, it's an ordinary PC with Linux

Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.115.0
Qt Version: 5.15.12
Kernel Version: 6.6.19-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 16 × AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics
Memory: 27.1 GiB of RAM
Graphics Processor: AMD Radeon Graphics
Manufacturer: LENOVO
Product Name: 21K4S02000
System Version: ThinkPad T14 Gen 4

chrfwow, 2024-07-07:
I am having a similar issue. Connecting to my Raspberry Pi Pico does not work. However, the port does not seem to remain occupied after I close VSCode, but I still cannot connect to its COM Port. 
When trying to connect to the COM Port from a Java application using jSSC, I get a Port Busy exception when VSCode is opened (as expected), but when VSCode is closed, I get the following error message:

```
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=21592, tid=22632
#
# JRE version: OpenJDK Runtime Environment (17.0+35) (build 17+35-2724)
# Java VM: OpenJDK 64-Bit Server VM (17+35-2724, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
```
I also cannot connect the Serial Monitor inside VSCode, or using rshell via WSL.

The errors occur both on my desktop pc and my laptop, with both running Windows 10

Prototype-Laboratory, 2024-11-19:
Partly it is working now.

If MicroPico isn't loaded at all, you can now enter REPL mode with the integrated serial monitor by using the terminal mode setting. Before, that was only possible outside VSCode, if I remember correctly.
[image]

If MicroPico is loaded and connected, it is not possible to open the COM-Port. This behavior is expected, as the port is occupied by MicroPico.
[image]

If I disconnect, I can open the Port, but REPL is not working, even if I Hard-Reset the Pico. It is only working, if the extension isn't loaded. So it seems, that the port will not totally freed, or the Pico itself is put into a strange state after disconnecting from the extension.
[image]

I am actually using the firmware 1.3.

> Basically I want to write a python program, which is connecting via pyserial to the Raspberry Pi Pico, which won't work, because the Port is still opened.

That statement was the original reason why I opened this issue. My workaround until now was, to just do not open the port and read and write directly to the port, which was somehow working. It is strange to not open or close a port and simply read and write to it, but I was able to share data through the serial connection. Unfortunately, sometimes the connections gets stuck and was not consistent.

I will check this later this week, how the behavior is now.

Prototype-Laboratory, 2024-11-27:
Ok after testing, the problem is still present. If I try to open the serial port with Serial.open() in python, a SerialException error is raised with "Port is already open.", even if the connection to the Pico by MicroPico is disabled. So something is still occupying the port.

It seems that this is a MicroPython related issue, as the problem exists after closing VSC and reconnect the Pico to the computer. So I have to use my workaround mentioned before.

@paulober

jkozniewski, 2025-01-23:
@paulober just wanted to report that after getting back to working on micropython projects again I have t osay I see a vast improvement in stability of the connection / uploading etc. It still occasionally get stuck after physically disconnecting and trying to connect again but it's happening much less than it used to + restarting  VS code and connecting again solves the issue without any additional steps ( like killing process manually and/or connecting via Thonny first)  

the only thing I have noticed is that now there it seems to be some delay when launching Run and also after Stop before it returns to REPL - guess maybe it's inevitable to provide more stability ? 

Anyway - thanks for your work on improving this extension !

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around connecting and reconnecting boards. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.

jkozniewski, 2026-09-14:
@paulober Haven't done much with pico lately - but I'll try to run some tests in within this month and report back. Thanks for continuous development, much appreciated!

closed 2026-10-22 by github-actions[bot]