paulober/MicroPico #209: Project configuration fails when creating new project on external drive

HaNaK0, 2024-03-12:
## What are the steps to reproduce this issue?
1. Create a new folder on the external drive
2. Open that folder in VS-Code
3. Run  `MicroPico: Configure Project`

## What happens?
I get this an error message and the operation seems to fail beacause no stubs or anything is added to the project nor will any of the autocompletion work.
[image]

## What were you expecting to happen?
I expect the stubs to be added to the project like when i run the same command on a folder that is on my machine

## Any logs, error output, etc?
``` 
2024-03-12 16:49:20.397 [error] Error: EISDIR: illegal operation on a directory, symlink 'C:\Users\satur\AppData\Roaming\Code\User\Pico-W-Stub' -> 'd:\Kod\test-micropico\.vscode\Pico-W-Stub' micropico.initialise {"value":"paulober.pico-w-go","_lower":"paulober.pico-w-go"} 
````

## Which version of MicroPico are you using?
v3.7.5

## Support info

```
Version: 1.87.1 (user setup)
Commit: 1e790d77f81672c49be070e04474901747115651
Date: 2024-03-06T00:21:36.258Z
Electron: 27.3.2
ElectronBuildId: 26836302
Chromium: 118.0.5993.159
Node.js: 18.17.1
V8: 11.8.172.18-electron.0
OS: Windows_NT x64 10.0.22631
```

paulober, 2024-03-14:
The issue is caused by the fact that the stubs are managet central and are only linked in your filesystem. Due to technical limitations, these links can't be created on external drives.
But I have a solution to this which I'll implement if I find time to do so.

HaNaK0, 2024-03-21:
Ok, i was guessing that was the issue, I like to have my code projects on my external drive so i'm looking forward to a solution. Might look into it myself if I get time and you accept help.

GregWoods, 2024-03-25:
First time using micropico, and I hot this bug straight away. I prefer to keep all my projects on a NAS. Switched to my local filesystem for now.

Josverl, 2024-04-01:
You can always add the stubs manualy if your preferences and MicroPico don't (yet) allign

closed 2024-11-18 by paulober