paulober/MicroPico #212: pico keeps disconnecting and reconnecting when i try to run a script on it

Emanuel-Tarcisiu, 2024-03-18:
hi everyone i have just bought my raspberry pico and i wanted to try and learn how to program a microcontroler and i didnt like the thonny ide(and couldnt get it to install micropython on my board) and i have found this vscode extension
(i have uploaded the uf2 file from the official website  for micropython)
once the pico connects i can run simple commands via the terminal like print() but once i try and use the run current file on pico command the board just disconnects and then reconnects almost instantly but my script isnt running

i only got his error message in the console: Exception occured (maybe a connection loss)

i dont know what am i doing wrong, i have also tried to connect my pico to my pc using a different usb cable but that one seems to do worse because it just kept connecting and disconnecting without running any commands(i am using a brand new micro-usb cable)

i also want to mention that i have run some simple c programs on my pico and they worked flawlessly
does anyone have a solution for this problem? thanks!
[image]
in this screen shot i ran a simple command and then tried to run a blink.py script on my pico but got that message

paulober, 2024-03-19:
Maybe that's related to issues some other also have.
Is It possible for you to upload a file via the virtual workspace function?

Emanuel-Tarcisiu, 2024-03-19:
i dont know how to do that where can i find this option? i used the options menu from when you right click the script to upload it.
[image]

i found a button that adds this remote workspace to my directory but i dont know how to use it
[image]

paulober, 2024-03-19:
Could can use It like a file system (it's on the pico). You can drag the python script in there and then right click the script and select run.

g0g2, 2024-03-22:
[image]

Hello, I am also experiencing the same issue. 
As instructed, I dragged the Python script and right-clicked on the script and select run, and the code worked as expected.

elf55, 2024-04-06:
I have the same problem.  Running MacOS 14.4.  Terminal cyclically connects and disconnects every 5 seconds.

Emanuel-Tarcisiu, 2024-04-06:
It worked! Dragging and dropping the file in the virtual workspace allowed me to run the script! Thank you very much! 
[image]

BojanKogoj, 2024-04-26:
This happens to me whenever it gets "stuck". Only I could fix it is by flashing it again.

poole-party, 2024-08-10:
I just ran into this as well on Ubuntu 22.04.4 LTS.

paulober, 2026-09-14:
This was reported against MicroPico 3.x, which used a connection backend that version 4 replaced completely. If you still see this on the latest version, please open a new issue. Thanks!

closed 2026-09-14 by paulober