paulober/MicroPico #213: Upload/ Sync syncFIleTypes seems to not respected when editing

ChrisChoke, 2024-04-08:
> Please provide a clear and concise description of the bug along with logs

my vscode project directory:

```
.
├── ESP32_GENERIC_C3-20240105-v1.22.1.bin
├── .micropico
├── README.md
├── src
│   ├── boot.py
│   ├── lib
│   │   ├── bme280_int.py
│   │   ├── microdot
│   │   │   ├── __init__.py
│   │   │   ├── microdot.py
│   │   │   └── utemplate.py
│   │   ├── mqtt_as.py
│   │   └── utemplate
│   │       ├── compiled.py
│   │       ├── recompile.py
│   │       └── source.py
│   ├── main.py
│   └── templates
│       └── index.tpl
├── test.py
├── update_libs.sh
└── .vscode
    ├── extensions.json
    ├── Pico-W-Stub -> /home/chris/.config/Code/User/MicroPico-Stubs/micropython-esp32-stubs==1.22.0.post1
    └── settings.json
```

settings.json:

```
{
    "python.linting.enabled": true,
    "python.languageServer": "Pylance",
    "python.analysis.typeCheckingMode": "basic",
    "python.analysis.diagnosticSeverityOverrides": {
        "reportMissingModuleSource": "none"
    },
    "micropico.syncFolder": "src",
    "micropico.openOnStart": true,
    "python.analysis.typeshedPaths": [
        ".vscode/Pico-W-Stub"
    ],
    "python.analysis.extraPaths": [
        ".vscode/Pico-W-Stub"
    ],
    "micropico.syncFileTypes": [
        "py",
        "txt",
        "log",
        "json",
        "xml",
        "html",
        "js",
        "css",
        "mpy",
        "tpl"
    ]
}
```
## What are the steps to reproduce this issue?
1. add 'tpl' filetype to syncFileTypes list
2. upload project to pico with crtl+p command
3. tpl file dont upload.
4. rename tpl to html
5. upload project to pico with crtl+p command
6. html file is uploaded to device

## What happens?

upload all files without my appended file types

## What were you expecting to happen?

expecting that my edited syncFileTypes list will be respected. it seems to  fall back to any default list or something.

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Which version of MicroPico are you using?

3.7.5

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
<paste here>
```

paulober, 2024-11-18:
Sorry for the delay. Can you confirm if this has been solved with v4.1.0?

ChrisChoke, 2025-03-12:
hey paul,

like in #214 said. its works. really great!!!

Thank you

Chris

closed 2025-03-12 by ChrisChoke