paulober/MicroPico #214: Sync Folder seems to not respected settings.json

ChrisChoke, 2024-04-08:
> Please provide a clear and concise description of the bug along with logs
my vscode project directory:

```
.
├── ESP32_GENERIC_C3-20240105-v1.22.1.bin
├── .micropico
├── README.md
├── src
│   ├── boot.py
│   ├── lib
│   │   ├── bme280_int.py
│   │   ├── microdot
│   │   │   ├── __init__.py
│   │   │   ├── microdot.py
│   │   │   └── utemplate.py
│   │   ├── mqtt_as.py
│   │   └── utemplate
│   │       ├── compiled.py
│   │       ├── recompile.py
│   │       └── source.py
│   ├── main.py
│   └── templates
│       └── index.tpl
├── test.py
├── update_libs.sh
└── .vscode
    ├── extensions.json
    ├── Pico-W-Stub -> /home/chris/.config/Code/User/MicroPico-Stubs/micropython-esp32-stubs==1.22.0.post1
    └── settings.json
```

settings.json:

```
{
    "python.linting.enabled": true,
    "python.languageServer": "Pylance",
    "python.analysis.typeCheckingMode": "basic",
    "python.analysis.diagnosticSeverityOverrides": {
        "reportMissingModuleSource": "none"
    },
    "micropico.syncFolder": "src",
    "micropico.openOnStart": true,
    "python.analysis.typeshedPaths": [
        ".vscode/Pico-W-Stub"
    ],
    "python.analysis.extraPaths": [
        ".vscode/Pico-W-Stub"
    ],
    "micropico.syncFileTypes": [
        "py",
        "txt",
        "log",
        "json",
        "xml",
        "html",
        "js",
        "css",
        "mpy",
        "tpl"
    ]
}
```
## What are the steps to reproduce this issue?
1. set syncFolder to "src"
2. upload project via ctrl+p command
3. toggle pico filesystem to verify the upload

## What happens?

all .py files on all relative pathes are uploaded to the board.

## What were you expecting to happen?

i excpect that the syncFolder perform a ```cd src```and upload all files within this folder and apply all rules at this stage.
i mean that synFileTypes applied at this folder and subfolder within it.
i excpect that the syncFolder will be at root on filesystem on my board. like if i do ```cp src/* /``` on linux.

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Which version of MicroPico are you using?

3.7.5

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
<paste here>
```

ChrisChoke, 2024-04-08:
i made a test and change the "default" part of the syncFolder configuration in my ```/home/chris/.vscode/extensions/paulober.pico-w-go-3.7.5-linux-x64/package.json```

if i do the setting there it work as excpected. it seems that the settings.json in the .vscode folder in my project folder do not respected. the defaults from package.json do override project setting or it just not loaded. i dont know. same with  #213 

Chris

webdeb, 2024-04-08:
Don't know if this is related, but sync config does not work in multi-folder workspaces. Should I create a new issue?

ChrisChoke, 2024-04-08:
yes that is exactly what happen.
if i toggle the "Toggle Pico-W-FS" button, the workspace change to a multi-folder-workspace. and all the settings in .vscode/settings.json dont work anymore. in that case we need a ..code-workspace file with all the config.

i think the workspace should change back to single workspace after toggling the pico-filsystem button again.

Chris

SmallhillCZ, 2024-06-04:
This is more of how VSCode extension settings work. IMO this setting should be instead in the `.micropico` file.

However since we have it in the extension setting for now, what you can do is save the VSCode workspace and set the extension settings there:

`./myproject.code-workspace`
```json
{
	"folders": [
		{
			"path": "."
		},
		{
			"name": "Pico (W) Remote Workspace",
			"uri": "pico:"
		}
	],
	"settings": {
		"python.linting.enabled": true,
		"python.languageServer": "Pylance",
		"python.analysis.typeCheckingMode": "basic",
		"python.analysis.typeshedPaths": [
			".vscode\\Pico-W-Stub",
			"~\\.micropico-stubs\\included"
		],
		"python.analysis.extraPaths": [
			".vscode\\Pico-W-Stub\\stubs",
			"~\\.micropico-stubs\\included"
		],
		"[python]": {
			"editor.defaultFormatter": "ms-python.black-formatter"
		},
		"python.analysis.diagnosticSeverityOverrides": {
			"reportMissingModuleSource": "none"
		},
		"micropico.syncFolder": "src",
		"micropico.openOnStart": true,
		"micropico.softResetAfterUpload": true
	}
}
```

Reinski, 2024-10-21:
What about the other issue mentioned by OP?
> i excpect that the syncFolder will be at root on filesystem on my board. like if i do `cp src/* /` on linux.  

The files from src folder should be copied directly into / and not to /src on the device. If the current behaviour was intended (i.e. specifying "src" will copy the whole folder), what about allowing to specify "src/*" as option to relatively create the copies on the device?  
I've seen a similar [issue](https://github.com/paulober/MicroPico/issues/167), however it is exact the exact opposite problem when uploading files outside the sync-folder, and since the OP mentioned my problem explicitely, I decided to add my question here.

paulober, 2024-11-18:
@webdeb Thanks for mentioning. Should be patched with the latest release. It was a scope issue of the settings.

@ChrisChoke I guess this should also be fixed with the latest release (v4.1.0), can you confirm?

xluthi, 2025-01-18:
Hello and thanks for this extension!

On my side, with release 4.2.1, putting micropico settings in my project `.vscode/settings.json` seems to not be taken into account (I've tried with `micropico.syncFileTypes`.). If I save my workspace and put the setting in the workspace configuration file, it is actually taken into account.

ChrisChoke, 2025-03-12:
Hey @paulober,

sorry for my huge delay. i have no good excuse for that.
i tested it right now with the current latest release. i can confirm. it works well. i am so happy. this take me on a new level while develop for those boards. 

Thank you that you made such a fantastic extension.

Chris