paulober/MicroPico #218: Under Linux, cannot connect ot Pico

davidbenucci, 2024-05-02:
Hi everyone. I'm using MicroPico underubuntustudio (kubuntu) 24.04 and I'm facing this issue: when I connect the pico I receive a notification of VS Code saying the connection is estabilished but in lower left corner the Pico remains disconnected and I cannot access to the Pico filesystem. I attach a screenshot of the problem. Reproducing the same steps in windows, everything is working fine.
Thanks in advance for your help

[image]
 
[image]

> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1.
2.
3.

## What happens?

## What were you expecting to happen?

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Which version of MicroPico are you using?

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
<paste here>
```

paulober, 2024-05-02:
Have you checked the permissions of your current user and the serial port? That's a common problem under various linux distros. Just google or look in the repos wiki to find a solution for your OS. If not please let me know.

paulober, 2024-05-16:
@davidbenucci Did this solve your issue?

davidbenucci, 2024-05-16:
Unfortunately not. I add my user to group dialout but without success.

elmanytas, 2025-04-16:
Hi @davidbenucci , I had a similar problem that was solved reflashing micropython.

Have you tryed the connection with thonny. IMHO thonny is worst that this extension but can help debugging connection problems to the microcontroller.

What is the microcontroller board you are using?

paulober, 2026-09-14:
I'm closing this as I couldn't get the details needed to look into it, and a lot has changed since. If it still happens with the latest version, please open a new issue using the bug form. Thanks!

closed 2026-09-14 by paulober