paulober/MicroPico #219: stubs for @rp2.asm_pio PIO methods

jkozniewski, 2024-05-13:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
Write code decorated with @rp2.asm_pio and using special functions for controlling PIO on Pico 

## What happens?
Pylance does not recognise those functions ( such as out(), wrap() etc ) and marks them as error (vie screenshot attached)
[image]

## What were you expecting to happen?
Pylance should be configured in a way that either disregards the contents of @rp2.asm_pio decorated function or should include definitions (stubs ?) of special functions used 

## Any other comments?
If there is any workaround that can be configured using Pylance settings directly (to somehow ignore a portion of code etc) I'd be grateful for tips of how to do so while waiting for some solution on extension-level (if possible at all).

## Which version of MicroPico are you using?
3.7.6

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.89.1 (Universal)
Commit: dc96b837cf6bb4af9cd736aa3af08cf8279f7685
Date: 2024-05-07T05:14:24.611Z
Electron: 28.2.8
ElectronBuildId: 27744544
Chromium: 120.0.6099.291
Node.js: 18.18.2
V8: 12.0.267.19-electron.0
OS: Darwin arm64 23.1.0
```

Josverl, 2024-05-14:
Hi @jkozniewski,

Can you please take a look at https://github.com/Josverl/PIO_ASM_typing and let me know if that would be helpful.

| With pio_asm_types |
| -------- |
| [image] |

I have not packaged or published this, So you will will need to copy some files into your typing folder, MicroPico puts that in a non default folder, but should not be  hard to find.

copy to  `typings/rp2`
-  https://github.com/Josverl/PIO_ASM_typing/tree/main/typings/rp2

include in your source ( and also synch the the rp2040 board) : 
- https://github.com/Josverl/PIO_ASM_typing/blob/main/src/typing_extensions.py

jkozniewski, 2024-05-25:
@Josverl Thanks for your contribution. I've just got back from a trip so I can test.
@paulober - so where is the folder for MicroPico typings where I can paste the files ?

paulober, 2024-05-25:
> @paulober - so where is the folder for MicroPico typings where I can paste the files ?

You can find the installed stubs in `$HOME/.micropico-stubs/included` (if you want to modify the included ones) but keep in mind that changes to this can be overwritten by a new update of the extension.)

jkozniewski, 2024-05-26:
Ok, thanks, I'll give it a try.

On a related note - seems that the micropython stubs are also not up to date in regard with `rp2.DMA` class and the attributes such as `count` etc - as per docs: https://docs.micropython.org/en/latest/library/rp2.DMA.html

Josverl, 2024-07-04:
@jkozniewski, 

Thanks for notification 

Currently I publish the stubs for each release, it may take some manual adjusting as new documentation tends to come with some quicks and additional adjustments that need hands on keyboard,
the stubs for v1.23.0 are available and these do include the DMA class
See : 
 - https://github.com/Josverl/micropython-stubs/blob/6b5416a269341331fce3dfe2e0f130373c78c1f5/publish/micropython-v1_23_0-rp2-stubs/rp2.pyi#L346-L470
 - https://github.com/paulober/MicroPico/discussions/230 and AFAIK this is now part of MicroPico.

I also build stubs for the preview versions of micropython - but I currently do no push them to PyPi as that would create a lot of churn.
If you need / want that they could be installed directly from Github with the proper URL 

If you find any other things that you´d like improved, please do let me know at : https://github.com/Josverl/micropython-stubs/issues

closed 2024-09-12 by paulober