paulober/MicroPico #223: wrapper_win32_amd64.exe detected as malware by Microsoft Defender on 1 Jun 2024

HLammers, 2024-06-01:
This morning, on the 1st of June 2024 Windows Defender on Windows 11 with all the latest automatic updates applied gave a 'severe thread blocked' message stating that Micropico's wrapper_win32_amd64.exe file was identified as trojan horse (I assume a false positive). Can this be resolved?

## What happens?
Windows Defender quarantined two copies of wrapper_win32_amd64.exe

## Any logs, error output, etc?
These are the two relevant Windows Defender reports:

> Detected: Trojan:Win32/Wacatac.B!ml
> Status: Quarantined
> Quarantined files are in a restricted area whare they can't harm your device. They will be removed automatically.
>
> Date: 01/06/2024 09:23
> Details: This program is dangerous and executes commands from an attacker.
>
> Affected items:
>    file: C:\Users\reg\.vscode\extensions\paulober.pico-w-go-3.7.7-win32-x64\dist\scripts\wrapper_win32\wrapper_win32_amd64.exe
>
>   process: pid:16264,ProcessStart:133617002161556784

> Detected: Trojan:Win32/Wacatac.B!ml
> Status: Quarantined
> Quarantined files are in a restricted area whare they can't harm your device. They will be removed automatically.
>
> Date: 01/06/2024 09:23
> Details: This program is dangerous and executes commands from an attacker.
>
> Affected items:
>    file: C:\Users\reg\.vscode\extensions\paulober.pico-w-go-3.7.7-win32-x64\dist\scripts\wrapper_win32\wrapper_win32_amd64.exe

## Any other comments?
I haven't checked yet if the functioning of Micropico is cripled

## Which version of MicroPico are you using?
v3.7.7

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.89.1 (user setup)
Commit: dc96b837cf6bb4af9cd736aa3af08cf8279f7685
Date: 2024-05-07T05:13:33.891Z
Electron: 28.2.8
ElectronBuildId: 27744544
Chromium: 120.0.6099.291
Node.js: 18.18.2
V8: 12.0.267.19-electron.0
OS: Windows_NT x64 10.0.22631
```

HLammers, 2024-06-01:
I can now confirm that VS Code doesn't see the Raspberry Pi Pico anymore when it's connected to the USB port. It works again if I undo the quarantined file.

Blargmode, 2024-06-01:
I have this issue as well. I downgraded to version 3.7.6 of the extension which works.

paulober, 2024-06-01:
Thanks for reporting. Unfortunately this is normal for every new update including `pyboard-serial-com` in the changelog at the moment. It takes some days until Microsoft adds the serial port wrapper to their list of false positives. Then everything should for normal if you refresh your AV definitions in MS Defender.

closed 2024-06-01 by paulober