paulober/MicroPico #225: Sync/Connection breaks on stopping multi-threaded program in Pico

SmallhillCZ, 2024-06-04:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Create a program that creates another thread on the second CPU core
2. Run the program, wait for the second core to start
3. Hit Stop in the status bar
4. Try to upload a file or refresh the Pico-W-FS

## What happens?
* Everything file sync related keeps spinning/loading forever, in REPL I can write but nothing gets executed,
* Pico gets disconnected and reconnects again, but this doesnt fix the issue, I have to restart VSCode

## What were you expecting to happen?
Stop the program, allow me to make a change and be able to start it again

## Any logs, error output, etc?
```ts
2024-06-04 10:57:41.731 [info] [Activator] Connection to wrapper successfully established
2024-06-04 10:59:11.288 [error] [Activator] Pyboard exited with code 1
2024-06-04 10:59:12.224 [info] [Activator] Connection to wrapper successfully established
```

## Any other comments?
* If my program doesn't get to the point where it runs an endless cycle on another core, and I stop it before, everything works OK.
* ⚠️ This also causes a great issue if I start the second thread in the main.py and try to connect to Pico, because it causes the same problem as the second thread starts before Pico is connected to the MicroPico extension. Fixed this by having five second wait at the beginning to allow connecting before.

## Which version of MicroPico are you using?
`v3.7.8`

## Support info

```
Version: 1.89.1 (user setup)
Commit: dc96b837cf6bb4af9cd736aa3af08cf8279f7685
Date: 2024-05-07T05:13:33.891Z
Electron: 28.2.8
ElectronBuildId: 27744544
Chromium: 120.0.6099.291
Node.js: 18.18.2
V8: 12.0.267.19-electron.0
OS: Windows_NT x64 10.0.26100
```

Mitch0S, 2024-06-20:
Having the same issue. It seems that when I duplicate the workspace, it reconnects correctly. Just putting this here to help with debugging. Maybe some sort of halt/stop code isn't being sent?

Mitch0S, 2024-06-20:
@SmallhillCZ on the microcontroller side, add a small delay (e.g. 3s) before creating the secondary thread. Helps quite a bit while we wait for this to be fixed.

anthonybrunelle, 2024-07-24:
I am also having this issue

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around stopping programs. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.