paulober/MicroPico #226: Pylance doesn’t recorgnise Micropython specific keywords anymore after reopening a project

HLammers, 2024-06-07:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Close a Micropython project (I opened a C++ project afterwards)
2. Open the project again

## What happens?
Pylance shows errors when using Micropython specific keywords. For example: `import machine` gives the error ‘Import "machine" could not be resolved Pylance [reportMissingImports]’.
The code can be uploaded to a Pico and runs normally.
 
## What were you expecting to happen?
I expect those Micropython specific keywords to be recognised normally after reopening the project, just as it used to be.

## Any logs, error output, etc?
```
2024-06-07 15:09:12.938 [info] Extension host with pid 10560 started
2024-06-07 15:09:12.939 [info] Skipping acquiring lock for c:\Users\reg\AppData\Roaming\Code\User\workspaceStorage\c9416504cbc29a5e9fea21234f2905d5.
2024-06-07 15:09:13.082 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onFileSystem:pico'
2024-06-07 15:09:13.100 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python', root cause: ms-python.debugpy
2024-06-07 15:09:13.160 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2024-06-07 15:09:13.209 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2024-06-07 15:09:13.231 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2024-06-07 15:09:13.432 [info] ExtensionService#_doActivateExtension ms-python.debugpy, startup: false, activationEvent: 'onLanguage:python'
2024-06-07 15:09:13.454 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2024-06-07 15:09:14.003 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2024-06-07 15:09:14.010 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2024-06-07 15:09:14.034 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2024-06-07 15:09:14.281 [info] Eager extensions activated
2024-06-07 15:09:14.296 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2024-06-07 15:09:14.300 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2024-06-07 15:09:56.685 [info] ExtensionService#_doActivateExtension vscode.extension-editing, startup: false, activationEvent: 'onLanguage:json'
2024-06-07 15:09:56.692 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:json'
2024-06-07 15:09:56.708 [info] ExtensionService#_doActivateExtension vscode.npm, startup: false, activationEvent: 'onLanguage:json'
2024-06-07 15:10:21.154 [info] Extension host terminating: renderer closed the MessagePort
2024-06-07 15:10:21.168 [error] Canceled: Canceled
	at new A (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:143:50615)
	at E.U (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:6007)
	at z.<computed>.M.charCodeAt.R.CharCode.DollarSign.z.<computed> (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:3091)
	at h.g (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:199937)
	at h.executeCommand (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:199375)
	at I.registerCommand.description (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:198699)
	at h (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:200476)
	at h.g (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:199468)
	at h.executeCommand (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:199375)
	at Object.executeCommand (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:163:22742)
	at exports.deactivate (c:\Users\reg\.vscode\extensions\paulober.pico-w-go-3.7.8-win32-x64\dist\extension.cjs:8:105588)
	at u.eb (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:159:9796)
	at c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:159:7867
	at Array.map (<anonymous>)
	at u.$ (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:159:7854)
	at u.terminate (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:159:8127)
	at f.terminate (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:163:1513)
	at t (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:176:10459)
	at MessagePortMain.<anonymous> (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:176:6675)
	at MessagePortMain.emit (node:events:514:28)
	at MessagePortMain._internalPort.emit (node:electron/js2c/utility_init:2:2285) setContext undefined
2024-06-07 15:10:21.169 [error] Canceled: Canceled
	at new A (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:143:50615)
	at E.U (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:6007)
	at z.<computed>.M.charCodeAt.R.CharCode.DollarSign.z.<computed> (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:150:3091)
	at h.g (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:199937)
	at h.executeCommand (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:199375)
	at I.registerCommand.description (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:198699)
	at h (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:200476)
	at h.g (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:199468)
	at h.executeCommand (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:153:199375)
	at Object.executeCommand (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:163:22742)
	at exports.deactivate (c:\Users\reg\.vscode\extensions\paulober.pico-w-go-3.7.8-win32-x64\dist\extension.cjs:8:105645)
	at u.eb (c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:159:9796)
	at c:\Users\reg\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:159:7867
[... 250 lines cut ...]
2024-06-07 15:29:45.493 [info] ExtensionService#_doActivateExtension ms-vscode.cmake-tools, startup: false, activationEvent: 'onTaskType:cmake'
2024-06-07 15:29:45.551 [info] ExtensionService#_doActivateExtension ms-vscode.cpptools, startup: false, activationEvent: 'onTaskType:cppbuild'
```

## Any other comments?
Rerunning Micropico: Configure project doesn’t resolve the issue, also not after deleting the .micropico file.

## Which version of MicroPico are you using?
3.7.8

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.90.0 (user setup)
Commit: 89de5a8d4d6205e5b11647eb6a74844ca23d2573
Date: 2024-06-04T19:33:54.889Z
Electron: 29.4.0
ElectronBuildId: 9593362
Chromium: 122.0.6261.156
Node.js: 20.9.0
V8: 12.2.281.27-electron.0
OS: Windows_NT x64 10.0.22631
```

paulober, 2024-06-09:
Does this also happen if you open a blank folder, run configure project and then create a main.py with import machine?

jimkr, 2024-06-09:
I'm encountering similar symptoms with a new project

> Does this also happen if you open a blank folder, run configure project and then create a main.py with import machine?

Yes, it does

Steps to reproduce on Windows 10:
1. Create an empty folder
2. In Visual Studio Code, select File | Open Folder, and select the newly created folder
3. Run `MicroPython: Configure Project`
4. Create a `main.p`y file with the single line `import machine`

At this point, I see the error from Pylance: `Import "machine" could not be resolved`

My `.vscode\settings.json` file contains this setting:
```
    "python.analysis.extraPaths": [
        "~\\.micropico-stubs\\included"
    ]
```

The micropico stubs do exist in my Windows user's home directory in a folder `.micropico-stubs\included`

As a workaround, I copied the .micropico-stubs directory from my home directory to my project directory.  I changed the above lines in `.vscode\settings.json` to also reference the project relative copy of the stubs:
```
    "python.analysis.extraPaths": [
        "~\\.micropico-stubs\\included",
        ".micropico-stubs\\included"
    ]
```
After saving `.vscode\settings.json`, the error resolving `import machine` goes away.

It appears that Pylance is not successfully using the stubs in my Windows user home directory.

HLammers, 2024-06-09:
> Does this also happen if you open a blank folder, run configure project and then create a main.py with import machine?

Yes, I can confirm that as well.
I haven’t tried @jimkr’s work around yet.

sandylnur, 2024-06-09:
I confirm the issue as well, @jimkr work around worked, instead of copying the whole directory I simply used the absolute path with my user account, for me being "C:\Users\<user-name>l\.micropico-stubs\included".

AleksanderKruecken, 2024-06-13:
> Does this also happen if you open a blank folder, run configure project and then create a main.py with import machine?

Yes

HLammers, 2024-06-14:
Following @sandylnur's suggestion to add the absolute path of the users folder to settings.json, partly resolves the problem, but not completely. It recognises `machine`, but it doesn't recognise parts of `gc` and `time`:
- `import gc` is recognised, so are `gc.enable()` and `gc.collect()`, but `gc.threshold()`, `gc.mem_free()` and `gc.mem_alloc()` are not.
- `import time` is recognised, but `time.sleep_ms()`, `time.ticks_ms()` and `time.ticks_diff()` are not.

jimkr, 2024-06-15:
>  it doesn't recognise parts of gc and time

I'd guess that pylance is using the python standard library versions of time and gc, which contain different methods.  It's not using the micropython stubs, so isn't aware of the micropython functions.

I'm not sure how (or whether it's even possible) to make the micropython stubs take precedence.

HLammers, 2024-06-15:
Could it have something to do with the version of MicroPython I'm using (1.23)? I don't think it should, because it used to be working when I was using the 1.23 preview.

paulober, 2024-06-16:
@HLammers No, the syntaxhighlighting and auto-completion is completely independent from the firmware installed on your board. The problem could be ~ as prefix for the user home in the vscode configuration. I have to do some test but I may have a solution to this.

whatdoido4, 2024-06-17:
Error in Python Language server output with absolute path entered into pylance settings "include" paths and nothing in settings.json:

2024-06-17 14:52:54.371 [info] Auto-Indent enabled
2024-06-17 14:52:54.377 [info] [Info  - 2:52:54 PM] (20844) Setting pythonPath for service "Firmware": "C:\Program Files (x86)\Python310-32\python.exe"
2024-06-17 14:52:54.377 [info] [Info  - 2:52:54 PM] (20844) Setting environmentName for service "Firmware": "3.10.11 (global)"
2024-06-17 14:52:54.576 [info] [Info  - 2:52:54 PM] (20844) Assuming Python version 3.10.11.final.0
2024-06-17 14:52:54.867 [info] [Info  - 2:52:54 PM] (20844) Found 143 source files
2024-06-17 14:52:54.908 [info] Auto-Indent enabled
2024-06-17 14:52:54.909 [info] [Info  - 2:52:54 PM] (20844) Setting environmentName for service "Pico (W) Remote Workspace": "3.10.11 (global)"
2024-06-17 14:52:55.166 [info] [Error - 2:52:55 PM] (20844) File or directory "pico:c%3A/users/#######/.micropico-stubs/included" does not exist.
2024-06-17 14:52:55.208 [info] [Info  - 2:52:55 PM] (20844) No source files found.
2024-06-17 14:52:55.214 [info] Auto-Indent enabled

commit 474c7dd4 (2024-06-22): Fix #226, Pylance doesn't work after reopening
PR #231 (2024-06-22): Patch v3.8.0
### Added
- Stubs auto-install feature to install selected stubs for a first opened project if not present

### Changed
- Upgraded to `pyboard-serial-com` `v3.1.0`,
- Updated dependencies
- Fixed #230, Update included stubs to v1.23.0 stable 
- Fixed #226, Pylance doesn't work after reopening
- Fixed #227, Switch stubs error (Thanks to @wtakuo for #228)
- Fixed #220, List remote fs contents and download project operations ignoring entries with spaces 
- Min VS Code version is now `v1.90.2`
- Updated stubs to MicroPython v1.23.0 stable (Thanks to @Josverl [`micropython-rp2-rpi_pico [... cut]
closed 2024-06-22 by paulober