paulober/MicroPico #234: Option to choose ports for manual connection

tintad, 2024-07-12:
# Feature Request 🚀
Add an option to choose ports for manual connection

## Is your feature request related to a problem? Please describe.
The auto connection function keeps crashing and only the manual one works. But every time I want to specify the COM port, I have to go to Settings and reload the windows

## Describe the solution you'd like
Add an options so it's possible to choose the ports manually and fast

paulober, 2024-07-21:
Thanks for the suggestion. Sounds like a good addition, i'll add it if i find some time.

tintad, 2024-09-12:
> Does v4 solve the crashing issues of auto-connect for you?

Still doesn't seem to work for me. Seems like it get confused with multiple devices plugged in my computer. It also pops up with this error

[image]

paulober, 2026-09-14:
Fixed in [MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0): you can choose the port with `micropico.manualComDevice` or `MicroPico: Switch Board`, and auto-connect no longer starts overlapping reconnect attempts. Thanks for reporting!

closed 2026-09-14 by paulober