paulober/MicroPico #239: Files not updated until a reconnect

Dejv311, 2024-07-26:
I have two files  _servo_test.py_ and a supporting file _servo_lib.py_ (that is being imported by  _servo_test.py_). Updates to the _servo_lib.py_ are not being reflected when running _servo_test.py_ until I reconnect the board.

**--- servo_test.py ---**
```
from servo_lib import Servo
import time
import math

sg90_servo = Servo(pin=14)  #To be changed according to the pin used

while True:
    #sg90_servo.move(0)
    coef = (math.sin(time.ticks_ms()/5000)+1)/2
    print("HELLO", coef)
    sg90_servo.move(90*coef)
    time.sleep_ms(20)
```

**--- servo_lib.py ---**
```
from machine import Pin, PWM

class Servo:
    __servo_pwm_freq = 50
    __min_u16_duty = 1640 - 2 # offset for correction
    __max_u16_duty = 7864 - 0  # offset for correction
    min_angle = 0
    max_angle = 180
    current_angle = 0.001

    def __init__(self, pin):
        self.__initialise(pin)

    def update_settings(self, servo_pwm_freq, min_u16_duty, max_u16_duty, min_angle, max_angle, pin):
        self.__servo_pwm_freq = servo_pwm_freq
        self.__min_u16_duty = min_u16_duty
        self.__max_u16_duty = max_u16_duty
        self.min_angle = min_angle
        self.max_angle = max_angle
        self.__initialise(pin)

    def move(self, angle):
        # round to 2 decimal places, so we have a chance of reducing unwanted servo adjustments
        angle = round(angle, 2)
        # do we need to move?
        if angle == self.current_angle:
            return
        self.current_angle = angle
        # calculate the new duty cycle and move the motor
        duty_u16 = self.__angle_to_u16_duty(angle)
        self.__motor.duty_u16(duty_u16)

    def __angle_to_u16_duty(self, angle):
        return int((angle - self.min_angle) * self.__angle_conversion_factor) + self.__min_u16_duty

    def __initialise(self, pin):
        self.current_angle = -0.001
        self.__angle_conversion_factor = (self.__max_u16_duty - self.__min_u16_duty) / (self.max_angle - self.min_angle)
        self.__motor = PWM(Pin(pin))
        self.__motor.freq(self.__servo_pwm_freq)
```

## What are the steps to reproduce this issue?
1. VS Code + MicroPico + all requirements
2. RP2040-Zero with RPI_PICO-20240602-v1.23.0 firmware
3. Changing the launch file _servo_test.py_ and hitting **Run** works fine
4. Changing the _servo_lib.py_ file (imported by _servo_test.py_) does not reflect the changes unless I unplug and plug back the RP2040

## What happens?
Changes to the file that is not being run are not reflected until the device (RP2040) is disconnected and reconnected (in SW using the status bar icon or physically).
After the device is reconnected and I hit **Run** on the _servo_test.py_, the changes in _servo_lib.py_ magically "appear" (as long as the changes were uploaded to PICO before disconnect).

## What were you expecting to happen?
The plugin should ideally auto-detect changes in _servo_lib.py_ and upload it when running _servo_test.py_.
I tried all 3 options (**Run current file on Pico**, **Upload current file to Pico**, **Upload project to Pico**) with the _servo_lib.py_ file, but running the _servo_test.py_ didn't reflect these changes.
Reconnecting the RP2040 made the changes in _servo_lib.py_ magically appear after hitting **Run** on servo_test.py (I did even not manually upload or run the servo_lib.py after reconnecting)

## Any logs, error output, etc?
2024-07-26 12:28:39.571 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onFileSystem:pico'
** nothing else **

## Which version of MicroPico are you using?
v3.8.2

## Support info

```
Version: 1.90.1
Commit: 611f9bfce64f25108829dd295f54a6894e87339d
Date: 2024-06-11T21:02:43.666Z
Electron: 29.4.0
ElectronBuildId: 9593362
Chromium: 122.0.6261.156
Node.js: 20.9.0
V8: 12.2.281.27-electron.0
OS: Linux x64 6.9.3-76060903-generic
```

paulober, 2024-09-11:
Are you working on the Pico directly (in the virtual filesystem)?

ziesemer, 2024-09-16:
I'm seeing the same.  The virtual filesystem appears to reflect the updated changes, but they are not reflected in the execution.

Dejv311, 2024-09-16:
@paulober 
> Are you working on the Pico directly (in the virtual filesystem)?

I am working on my PC, then uploading the files to Pico

paulober, 2024-09-16:
So detecting dependencies isn't supported at the moment. So if you have two files, you need to make sure to upload them both before executing anything.
Also i don't relay understand what you mean by magically appears as the extension doesn't upload anything to the pico on it's own. One reason i could see beeing an issue here is that if your files get run on the pico tha class servo is stored in memory. So if you then rerun the servo_lib by importing it, it will fail to create the new version of the class as already one with this name exists.
We can see if this is realy the problem if you run `del Servo` after executing your stuff, then changing anything and doing just the normal workflow to reproduce the issue but for executing a second time.

paulober, 2024-09-16:
Just implemented soft-resets without follow as default before and after file executions.
This should fix this issue for you @Dejv311 @ziesemer 
Can you confirm?

commit 9a4fb65f (2024-09-16): Fix #239, Soft-resets before and after file run
closed 2024-09-16 by paulober