paulober/MicroPico #240: Can not execute code from PC

ToMiDevelop, 2024-08-04:
## Brief error overview

I'm having an issue with running code from PC on the board. It ise very strange behaviour.

## Details of the error

I'm having and already partly developed project (you can see it's structure in the file explorer picture attached bellow).

[image]

When I'm clicking Run button when having **main.py** opened I receive an arror in the console:

Exception occured (maybe a connection loss)

The strange thing is that when I run the same **main.py** saved on the Pico board everything seems to work fine.

## Additional information

This behaviour is not observed in other MicroPython editor - tried Thonny. Over there everything seems to work fine - even running script located physicaly on the PC.

The problem started to occur when Widnows (I use 11) updated itslef and some usb drivers.
After this there were even no messages beeing caught in the serial console. Now after using Zadig and updating the driver (with board plugged in boot mode) I am able to see all the serial print() messages, but still not able to run scripts storred on the PC.

Maybe this is the case? I'm pointing this as a Bug, becasue probably more people see the same issue.

paulober, 2024-09-11:
@ToMiDevelop Has the issue been fixed with v4.0.0+?

ToMiDevelop, 2024-09-21:
> @ToMiDevelop Has the issue been fixed with v4.0.0+?

Tested on 4.0.6

ESP32 WROOM board works, RP2040 based board works.

paulober, 2024-09-23:
Great to hear that, so I can closed this issue now, or do you have any other issues?

ToMiDevelop, 2024-09-23:
Could you still have this issue open for few days? Currently I'm preparing a custom MicroPython build tailored for ESP32 C6 based board and would like to check this board also. Probably tomorrow I'll have results and post them.

ToMiDevelop, 2024-09-29:
> Could you still have this issue open for few days? Currently I'm preparing a custom MicroPython build tailored for ESP32 C6 based board and would like to check this board also. Probably tomorrow I'll have results and post them.

Tested and works even on self tailored builds of MicroPython. Thread can be closed.

closed 2024-09-29 by paulober

Lemaya, 2025-05-04:
I also have this issue with Version 4.2.2
The connection to the board is established, but I cannot run code nor upload files

2025-05-04 12:01:41.002 [warning] [Activator] Failed to execute script on Pico.

[image]