paulober/MicroPico #244: Does not start due to wrapper_linux_amd64.bin lacking executable permission

DanielJutz, 2024-08-28:
## What are the steps to reproduce this issue?
1. using visual-studio-code-bin on arch linux
2. install micropico
3. open folder
4. micropico: configure project

## What happens?
Seemingly nothing. Project doesn't get configured.

## What were you expecting to happen?
Project to be configured.

## Any logs, error output, etc?
"Developer: Show Logs... > Extension host" shows following error
```
2024-08-28 23:05:53.280 [error] Error: spawn /home/daniel/.vscode/extensions/paulober.pico-w-go-3.8.3-linux-x64/dist/scripts/wrapper_linux_amd64/wrapper_linux_amd64.bin EACCES
	at ChildProcess._handle.onexit (node:internal/child_process:286:19)
	at onErrorNT (node:internal/child_process:484:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
```

## Any other comments?
executing
`
chmod +x /home/daniel/.vscode/extensions/paulober.pico-w-go-3.8.3-linux-x64/dist/scripts/wrapper_linux_amd64/wrapper_linux_amd64.bin
`
fixes the issue. I am curious if any other files are lacking permissions.

## Which version of MicroPico are you using?
3.8.3

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.92.2
Commit: fee1edb8d6d72a0ddff41e5f71a671c23ed924b9
Date: 2024-08-14T17:29:30.058Z
Electron: 30.1.2
ElectronBuildId: 9870757
Chromium: 124.0.6367.243
Node.js: 20.14.0
V8: 12.4.254.20-electron.0
OS: Linux x64 6.10.6-arch1-1
```

kellyjp, 2024-08-29:
I am experiencing the same issue on MacOS.
Adding execute permission to ~/.vscode/extensions/paulober.pico-w-go-3.8.3-darwin-arm64/dist/scripts/wrapper_macOS_arm64/wrapper_macOS_arm64.bin got things working again for me.
This appears to be a regression in 3.8.3 which appears to have automatically installed on my Mac last night.

paulober, 2024-08-29:
Sorry for that guys. I'll try to publish a package with the right permissions ASAP.

PR #245 (2024-08-29): Patch v3.8.4
### Added

- Eperimental support for `ESP32-S3-Pico`

### Changed

- Upraded to `pyboard-serial-com` `v3.1.2`
- Updated dependencies
- Ignore `.idea` by default (Thanks to @Finnomator for #237) 

(Fixes #244 )
closed 2024-08-29 by paulober