paulober/MicroPico #246: Cant connect to PICO-W with latest update

brianreinhold, 2024-08-31:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. install latest update
2.
3.

## What happens?
No connection to Pico (W) REPL>>>

## What were you expecting to happen?
connect to PICO-W

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?
Had to revert back to version prior to V3.8.4

## Which version of MicroPico are you using?
v3.8.2

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.92.2 (user setup)
Commit: fee1edb8d6d72a0ddff41e5f71a671c23ed924b9
Date: 2024-08-14T17:29:30.058Z
Electron: 30.1.2
ElectronBuildId: 9870757
Chromium: 124.0.6367.243
Node.js: 20.14.0
V8: 12.4.254.20-electron.0
OS: Windows_NT x64 10.0.19045
```

paulober, 2024-09-01:
please try v3.8.4

brianreinhold, 2024-09-01:
That is the version I got (automatically) and it refused to connect to the PICO-W. I downgraded to v3.8.2 and I got the connection back.

brianreinhold, 2024-09-01:
Yes I have. That was the first thing I did as I thought it was a VS Code config problem. I have my PICO-W working again and really need it to work at this time. I may try it again perhaps after a new revision

paulober, 2024-09-01:
@brianreinhold Understand. There'll be a new release come in the next days with a much more stable backend which should fix many of these stability issues. Hope 3.8.2 is working for you till it get's released.

brianreinhold, 2024-09-11:
I tried both 4.0.0 and 4.0.2. They work partially. When I first open VS and the PICO is connected, I can click on the pico disconnected text at the bottom and it will connect. However, if I unplug the PICO and plug it back in, I can never get VS to reconnect. I have to exit VS code and restart.

So I need to revert back to 3.8.2 as it is very important to be able to click on reset (if I edit some files) followed by an unplug/replug of the PICO in order for the (Bluetooth) pairing keys to be generated. I do not understand why that is needed but it has been something I have noted since first getting pairing to work in Micropython.

This is not a problem if I have the PICO run automatically on a plugin event without VS Code. Pairing always comes up.

Milan0x4d, 2025-01-02:
Thank you for this thread Brian. 

I just started with microcontrollers in general last night, after I bought my first pico. 

I established the terminal connection (via 'minicom') to REPL from Fedora fairly quickly. Then I tried to get the  VS Code to work and I started getting "No COM device found" errors.

Your suggestion to downgrade to v3.8.2 fixed all the issues and I have a REPL console from VS Code!

Prior to this, I followed the recommendations from this project page:
https://github.com/paulober/MicroPico/wiki/Linux

Of the two solutions, I first chose to add my user to the group 'dialout'. This group owns the /dev/ttyACM0 on my machine.
After I did this, I no longer had to run minicom with 'sudo'. It just worked. This proved that the user has access to the device. 

However, even after the logout/restart, VS Code would not work. It's as if it doesn't care about the group membership of the account under which it runs. Could this perhaps be the issue?

brianreinhold, 2025-01-02:
@Milan0x4d I am running on Windows so I really can't comment about the various permission configurations in Linux. I really did not go any further than downgrading as it worked. I have tried a few of the latest updates but they continue to have the same problem.

germanstafe, 2025-03-12:
@brianreinhold  Hi, could you please tell me how to downgrade the Pico 4? I'm desperate. I can't connect since the update.

brianreinhold, 2025-03-12:
@germanstafe I assume you mean the VS Code extension and not the PICO firmware version.
Navigate to the Micropico extension as if you were going to install it.
I have three options because I have version 3.8.2 installed
`update to v4.2.1`  `Disable`  `uninstall`

Believe it or not, click the dropdown carat in the uninstall option and you will have two choices: _uninstall_ and _install specific version_
Click the _install specific version_ option and another huge dropdown list will appear with all the different versions. The latest one I found that works is 3.8.2. I have not tried the current version but I assume from your comment it still does not work.

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around connecting and reconnecting boards. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.

brianreinhold, 2026-09-14:
FYI: I originally opened this issue but am now retired and the company I worked for no longer exists. So I will be unable to participate (assuming the issue still exists).